ArduPilot/ardupilot_wiki #233: Create Xracer build instructions in dev wiki

hamishwillee, 2015-11-29:
New feature [discussed here](https://groups.google.com/forum/#!msg/drones-discuss/IUlLAKVZBr0/JpimC_hdCQAJ).

rmackay9, 2015-12-01:
let's put a video in there like this one (until a better one arrives): https://www.youtube.com/watch?v=YNlDeSOwWi0

rmackay9, 2015-12-01:
we can also mention that the reason that it's smaller is that it is different from the Pixhawk in that it does not have the I/O chip on it.  This also slightly reduces lag between the control and pwm outputs.

davidbuzz, 2016-11-15:
AFAIK the Xracer was released as the "pixracer", which is now well documented, so this issue can be closed.  see https://github.com/ArduPilot/ardupilot_wiki/search?utf8=%E2%9C%93&q=pixracer

closed 2016-11-15 by davidbuzz