ArduPilot/ardupilot_wiki #234: ROS+ardupilot: sim_vehicle.sh: command not found

CrazyPopLin, 2015-12-02:
Hi, I am following [Using ROS/Gazebo Simulator with SITL](http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/). Also, I've checked #194 .
I git ArduPilot and I did follow steps one by one to install each ros packages and build them without errors. 
Also I source **setup.bash** and **build path** and ouput **echo $PATH** below:
`/home/lin/catkin_ws/devel/bin:/opt/ros/indigo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:$/home/lin/ardupilot/Tools/autotest`.
Then 
`cd ~/ardupilot/ArduCopter`
`sim_vehicle.sh -f arducopter_sitl_ros --console`
and get **sim_vehicle.sh: command not found**.

Any suggestions will be appreciate. Thanks.
I also have another small problem which is why need `cd ~/ardupilot/ArduCopter` to run `sim_vhicle.sh command`, the **ArduCopter** just has some cpp files and head files.

hamishwillee, 2015-12-02:
Is there really a $ in the path as shown:
```
...games:$/home/lin/ardupilot/Tools/autotest
```
I would expect it to look like this:
```
...games:/home/ubuntu/ardupilot/Tools/autotest
```
So you're getting the issue because the path isn't actually pointing to /Tools/autotest where **sim_vehicle.sh** is located.

Please fix and confirm this worked.

CrazyPopLin, 2015-12-02:
thx, it works.

closed 2015-12-02 by CrazyPopLin