ArduPilot/ardupilot_wiki #236: make sitl  error on Cygwin

uri200, 2015-12-04:
Hello,
I am trying to build AurduCopter at CygWin and I am getting ArduCopter.elf' failed when i try toi execute make sitl -j4. Any fix for this issue?
Thanks!!

C:/cygwin/home/uri/ardupilot/mk/board_native.mk:106: recipe for target '/tmp/ArduCopter.build/ArduCopter.elf' failed
make: *** [/tmp/ArduCopter.build/ArduCopter.elf] Error 1

hamishwillee, 2015-12-04:
That looks like a "real" error in the build rather than an error in the docs for the build process. Can you please re-raise on the APM forums or in the ardupilot (source code) wiki.

I am closing this. Thanks for taking the time to post!

closed 2015-12-04 by hamishwillee

uri200, 2015-12-04:
Thanks. 
I did 
http://ardupilot.com/forum/viewtopic.php?f=69&t=14489&sid=1b16920190f4cc17e55c0873cf6c02a4