ArduPilot/ardupilot_wiki #239: Use of servo or DC stepper motors with no position feedback

raptortrias, 2015-12-07:
Hi,

 Is the use of motors with no position feedback supported in antenna tracker. Since the pixhawk has all the sensors it should be possible to use it for orientation of the antenna. I am looking to integrate heavy duty stepper motors that lack position feedback

[Configuration](http://antennatracker.ardupilot.com/wiki/configuration/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

hamishwillee, 2015-12-07:
Hi @raptortrias 

This tracker is for issues with the documentation. Please raise this technical support query on the APM forums (link at bottom of any wiki page)

Regards
Hamish

closed 2015-12-07 by hamishwillee

rmackay9, 2015-12-07:
@raptortrias, yes, that should work i think.