ArduPilot/ardupilot_wiki #241: Use Git ,Not Github ! Building ArduPilot for Pixhawk/PX4 on Windows with Make

whyscience, 2015-12-10:
[Building ArduPilot for Pixhawk/PX4 on Windows with Make](http://dev.ardupilot.com/wiki/building-px4-with-make/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

[image]
 

If You use Github, there will be a lot problems .
But use Git ,everything is OK !

hamishwillee, 2015-12-10:
I'm sorry, I don't understand what the problem is here/solution that needs to be applied to the documents. Can you explain more?

whyscience, 2015-12-11:
Maybe I just got wrong
When I plug in Jlink ,problems still happen

hamishwillee, 2015-12-11:
I'm still confused - have no idea what you are suggesting.

whyscience, 2015-12-11:
When I use github to clone the code,build the code with Eclipse there will be errors 
But with Git ,errors  are gone.
However when I plug in Jlink ,errors still happen

hamishwillee, 2015-12-11:
Github is a repository management system that uses the git tool for version control, so you're always using git in some way to get the code. I think what you're saying is that the Github for Windows tool properly syncs the repo but you get build errors if you use some other git version. I suspect that perhaps you've got line endings or something set differently between the two versions of the tool.

Either way, I do not believe there is any issue with this documentation. If you follow it as written, the code will build. Thanks!

closed 2015-12-11 by hamishwillee