ArduPilot/ardupilot_wiki #251: Update" Using ROS/Gazebo Simulator with SITL"

hamishwillee, 2015-12-22:
update [Using ROS/Gazebo Simulator with SITL](http://dev.ardupilot.com/wiki/using-rosgazebo-simulator-with-sitl/) with new information from the team who created [New ArduPilot plugin for Gazebo](http://diydrones.com/profiles/blogs/705844:BlogPost:2151758?id=705844%3ABlogPost%3A2151758&page=-1)

Contact Aurelien ROY and co with reminder if this does not happen automatically in the new year. They will also need logins sorted out.

doctorroboto, 2016-03-16:
I am having no luck following the instructions posted on this page. In addition to missing a few small packages, these steps say nothing about installing Gazebo, which I believe is part of the problem. If I try to step through this, eventually I hang up when running this step:

rosdep install --from-paths src --ignore-src --rosdistro indigo -y

This fails with the error:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
glog_catkin: Cannot locate rosdep definition for [gflags_catkin]

I assumed (perhaps incorrectly) that I should be installing Gazebo6 following the instructions here: http://gazebosim.org/tutorials?cat=install&tut=install_ubuntu&ver=6.0. However, given that some other instructions point specifically to a build of Gazebo that's already linked to ROS (http://gazebosim.org/tutorials?tut=ros_wrapper_versions#Gazebo6.xseries, which is linked from http://dev.px4.io/simulation-gazebo.html), I have no idea what to do.

Can someone help me figure out what is wrong with the install, and then update the installation instructions on the Ardupilot page to contain the correct steps?

Frechetta, 2016-04-07:
I've found that cloning the [gflags_catkin](https://github.com/ethz-asl/gflags_catkin) repository into the src directory fixes the issue.

auturgy, 2017-10-15:
Closed as per @khancyr comment above.

closed 2017-10-15 by auturgy