ArduPilot/ardupilot_wiki #255: Flying with a Joystick/Gamepad

clemtrinc, 2015-12-28:
[Flying with a Joystick/Gamepad](http://copter.ardupilot.com/wiki/flying-with-a-joystickgamepad-instead-of-rc-controller/) - No failover to transmitter if the usb joystick is disconnected from the usb port, or if I cut the telemetry link, I followed the exact procedure from the page, this was tested 6 motnhs ago, and again today after 2 updates of mission planner, on both a apm and pixhawk.

hamishwillee, 2015-12-28:
I *think* that the instructions are correct, so if this doesn't work it is probably a bug in ardupilot itself. I have therefore re-posted as https://github.com/diydrones/ardupilot/issues/3389

closed 2015-12-28 by hamishwillee

clemtrinc, 2015-12-28:
Alright thanks, sorry for the mistake I'm new to this community

hamishwillee, 2015-12-29:
@clemtrinc Seriously, not a problem. It is possible this is a documentation problem - but the core team will know for sure.

CraigElder, 2015-12-29:
@clemtrinc Please post a log file showing the issue. Thanks

clemtrinc, 2015-12-29:
Alright, I can't do this until next week as I'm out of town, but I'll do
this first thing.