ArduPilot/ardupilot_wiki #256: 3DR Solo Example to change to Mode1 is incorrect

mtbsteve, 2015-12-28:
[3DR Solo (Developer Information)](http://dev.ardupilot.com/wiki/solo/)
Hi,
I just went through the example provided to change the stick mapping to mode1.
The example actually describes the settings for mode4, having throttle and roll on the left stick and pitch and yaw on the right stick. 
The correct mapping for mode 1 should be:
throttle stick-id = 1, dir = 0
roll stick-id = 2, dir = 0
pitch stick-id = 3, dir = 1
yaw stick-id = 0, dir = 0
Thx Stephan

hamishwillee, 2015-12-29:
Thank you. I've changed the text as you suggested (I haven't attempted to verify it though).

hamishwillee, 2015-12-29:
Hi @mtbsteve Have you actually checked the instructions?

hamishwillee, 2015-12-29:
Actually, I have deleted this section because it is possible to change modes using the Solo Android app. I don't see any real benefit on providing programmatic access through the wiki to do this ... especially as it sounds like this interface may change (or have changed). Thanks for your help. Closing.

mtbsteve, 2015-12-29:
@hamishwillee I don't agree to remove the documentation. The interface is still valid for FW1.3. If it changes in future, adapt it then accordingly. Yes with the Android app you can switch between Mode1 and 2, but there's no way to access other stick modes like 3 and 4. 
For me as a long standing Mode 3 RC pilot the missing stick mode configuration was always an issue with Solo until I found this wiki page. 
So I would much appreciate if 3DR may reconsider and put in a proper documentation - or even better implement it into both the iOS and Android App. Thanks!

hamishwillee, 2015-12-29:
Hi @mtbsteve 

This is a DroneCode wiki - its not up to 3DR to decide what goes here or not :-) We all decide, and ultimately if there is disagreement the wiki owner does.

So as you've verified the information as still valid I've restored it. 

Can you provide instructions on how to access the other stick modes so we can extend it?

> or even better implement it into both the iOS and Android App.

I will ask :-)

hamishwillee, 2015-12-29:
@mtbsteve There is an intent to add support for different modes through the controller, but this is being prioritised against all the other requirements. Should happen eventually!

mtbsteve, 2016-01-01:
@hamishwillee thanks for following up.
Here are the stick mappings for Mode 3 and Mode 4:
The mapping for mode 3:
throttle stick-id = 1, dir = 0
roll stick-id = 0, dir = 0
pitch stick-id = 3, dir = 1
yaw stick-id = 2, dir = 0

The mapping for mode 4:
throttle stick-id = 3, dir = 0
roll stick-id = 0, dir = 0
pitch stick-id = 1, dir = 1
yaw stick-id = 2, dir = 0
Thx Stephan

hamishwillee, 2016-01-03:
Hi @mtbsteve 

Thanks very much. I've integrated it [here](http://dev.ardupilot.com/wiki/solo/#changing_the_controller_mode). No settings for Mode 2?

auturgy, 2017-10-15:
@Pedals2Paddles can you have a look at this and let us know if this can be closed (or close it)?
Thanks.

brunoolivieri, 2019-07-16:
Do we have newer information about this? Make sense maintain this issue open?

rmackay9, 2019-07-16:
Let's close it..

closed 2019-07-16 by rmackay9