ArduPilot/ardupilot_wiki #260: common-Pixhawk Overview

lvale, 2016-01-08:
The last sentence of this paragraph about Virtual Pin 103 is not correct, because SBUS out is now implemented and controlled by a parameter 

http://copter.ardupilot.com/wiki/arducopter-parameters/#enable_use_of_sbus_output_brd_sbus_out

Virtual Pin 103: RSSI (Received Signal Strength Input) input pin voltage (SBus connector output pin). This is the voltage measured by the RSSI input pin on the SBUS-out connector (the bottom pin of the 2nd last servo connector on the 14 connector servo rail). **Can alternatively serve as SBus out but not yet implemented.**

hamishwillee, 2016-01-19:
Thanks @lvale. I've updated [the text](http://planner.ardupilot.com/wiki/common-pixhawk-overview/#pixhawk_analog_input_pins) to look like this - OK?

[image]

lvale, 2016-01-19:
Perfect @hamishwillee 
Thanks

closed 2016-01-19 by lvale