ArduPilot/ardupilot_wiki #261: Mission Planner Language Translations

robetyosua, 2016-01-08:
i just finished translate into bahasa indonesia, i hope you can implement this language on mission planner. thank you
[translation.zip](https://github.com/diydrones/ardupilot-wiki-issue-tracker/files/82799/translation.zip)

hamishwillee, 2016-01-19:
Hi @robetyosua 

Thanks very much. Can you please repost these on the Mission Planner issue tracker here: https://github.com/diydrones/MissionPlanner/issues ?

This area is for issues with the wiki and its content.

I am closing.

Regards
Hamish

closed 2016-01-19 by hamishwillee

robetyosua, 2016-01-25:
ok sorry @hamishwillee  , I 've repost my request here : https://github.com/diydrones/MissionPlanner/issues 
Thanks @hamishwillee 

regards,

Robet

hamishwillee, 2016-01-25:
@robetyosua  No need to be sorry - thanks for your contribution :-) !!!