ArduPilot/ardupilot_wiki #262: Setting up SITL on Linux. Installing packages

approximator, 2016-01-12:
[Setting up SITL on Linux](http://dev.ardupilot.com/wiki/setting-up-sitl-on-linux/)

In the section: **JSBSim (Plane only)** stated that *libtool* should be installed. However, on my Ubuntu 15.10 command 
```
./autogen.sh --enable-libraries
```
 outputs an error 
```
**Error**: You must have `libtool' installed to compile jsbsim.
```

I suspect that package **libtool-bin** should be installed on Ubuntu.

hamishwillee, 2016-01-19:
@approximator According to the preceding line you need to have installed libtool:

> sudo apt-get install libtool automake autoconf libexpat1-dev

Please confirm you did this first? 

... and that libtool-bin was also necessary?

approximator, 2016-01-19:
Yep, it seems that I didn't install all the packages from the line above.
So, you are right it works now.

closed 2016-01-19 by approximator