ArduPilot/ardupilot_wiki #270: Cant work with Joystick

sellithy, 2016-01-18:
I opened the Webpage where it tells you how to operate APM with a joystick and followed it. I could setup controls. And can Calibrate. But when I go to the home screen it keeps repeating "PreArm: RC not calibrated."

Tried to change settings to low Latency, but then it stuck at Getting Params. I couldn't change MAVLINK to 2 because I didn't know where.

rmackay9, 2016-01-18:
The issue is that the joystick's range is exactly 1100 to 1900 and the pre-arm checks require that they be at least 1 pwm value off or it thinks the calibration hasn' been done.  If you go in manually and change RC1_MIN to 1101 and RC1_MAX to 1901 and then repeat for RC2_, RC3_ and RC4_ then it should work.

closed 2016-01-18 by rmackay9

zlite, 2016-01-18:
Maybe we should set those defaults to 1099 and 1901, or something like that?

sellithy, 2016-01-18:
A proof to the Drone Community is so small: I saw both of you in many different contexts about Drones. In all of those you helped me so much. Thank you

sellithy, 2016-01-18:
Probably a stupid queston, but when trying to do esc calibration i get "set param error. Please ensure your version is AC3.3"

zlite, 2016-01-18:
Are you using AC 3.3? If not, you should upgrade

hamishwillee, 2016-01-19:
@rmackay9 Is it reasonable that this behaviour be documented? I am assuming that the 1100 and 1900 values are joystick dependent?

At the end of the setup section (http://copter.ardupilot.com/wiki/flying-with-a-joystickgamepad-instead-of-rc-controller/#setup_with_the_mission_planner) we could have a note (if you think it useful?):

>  An error "PreArm: RC not calibrated" (following calibration) indicates that the Joystick PWM values lie exactly on the nearest "100" values (e.g. 1100 and 1900). The pre-arm checks assume that if the values are not at least 1 pwm off the century value, that calibration has not been done. 

>   If this is the case, you will need to manually change RC1_MIN to 1101 and RC1_MAX to 1901 and then repeat for RC2_, RC3_ and RC4_.

rmackay9, 2016-01-19:
@hamishwillee, I don't think it's joystick related actually.  I think the MP does some kind of internal mapping from whatever the joysticks return to the 1100 ~ 1900 pwm range.  So adding a note to the wiki would definitely help people, thanks for bringing this up!

hamishwillee, 2016-01-19:
Thanks @rmackay9 . Done - the following note was added:

[image]

Let me know if that text needs changing, otherwise I'll consider this issue closed.

rmackay9, 2016-01-19:
@hamishwillee, looking good, thanks!