ArduPilot/ardupilot_wiki #273: common-LightWare SF10 Lidar

RIGeek, 2016-01-19:
[common-LightWare SF10 Lidar](http://copter.ardupilot.com/wiki/common-lightware-sf10-lidar/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

I've set up the device with firmware 3.3.2 for Serial 4 as described in this document but receive a zero value for SonarRange in status. I've triple checked the wiring, baud, etc. I've tested my cable using jumpers connected to a USB to FTDI adapter and can see the range value via Putty. I've also tested my Serial 4 port by attaching the GPS to that port. 

I've worked with 3DR support and they verified that I did the above process and then let me know that they would be unable to help me any farther. I somehow feel as though the RNGFND values are not linking the device to Serial 4.

rmackay9, 2016-01-19:
This is not really the correct place to be reporting this issues I think but in any case, could you attach a dataflash log file?

hamishwillee, 2016-01-19:
@RIGeek Github is mostly for reporting documentation issues/errors. The right place for you to post this technical support query  - with logs - [is this forum](http://ardupilot.com/forum/viewforum.php?f=90). I am closing this issue here.

closed 2016-01-19 by hamishwillee

RIGeek, 2016-01-19:
OK. Sorry. I ended up working with someone that was able to help. There was a bug in the 3.3.2 firmware.

rmackay9, 2016-01-20:
@RIGeek, oh!  what bug was that?  I just want to make it easier for the next person.
Great that you got it working.

RIGeek, 2016-01-20:
I'm unsure of the details of it. I was suggested to use another revision firmware.

http://diydrones.com/forum/topics/lightware-sf10-c-and-pixhawk-ac-3-3-2-serial-connection#