ArduPilot/ardupilot_wiki #276: QGroundControl information is out of data

DonLakeFlyer, 2016-01-21:
[common-Choosing a Ground Station](http://copter.ardupilot.com/wiki/common-choosing-a-ground-station/) - 

Full support for ArduPilot firmware vehicles in QGroundControl is now in beta. This includes full Setup and Configuration, Mission Planning, Flight Display as well as streaming video. Also QGroundControl is now supported an Android Tablet.

The Beta link in here: http://www.qgroundcontrol.org/apm
Download link: http://www.qgroundcontrol.org/downloads
Wiki: Should link to qgrouncontrol.org no to a deeper page
Support forum: https://groups.google.com/forum/#!forum/qgroundcontrol
Gitter: https://gitter.im/mavlink/qgroundcontrol

hamishwillee, 2016-01-22:
Thanks @DonLakeFlyer! I've updated the page as you indicated!

closed 2016-01-22 by hamishwillee

DonLakeFlyer, 2016-01-22:
Awesome, big thanks Hamish

DonLakeFlyer, 2016-01-22:
Just realized I should have provided you with a new image as well. Here is a good one. Sorry about that.

[image]

hamishwillee, 2016-01-25:
Thanks Don - I added that now.