ArduPilot/ardupilot_wiki #278: The camera+gimbal landing page should have links to the ready-for-purchase camera trigger boards

rmackay9, 2016-01-26:
I think this page, http://copter.ardupilot.com/wiki/common-cameras-and-gimbals/, should have links to these other pages:
http://copter.ardupilot.com/wiki/common-camera-shutter-triggering-for-sony-multiport-connectors-using-seagull-map/
http://copter.ardupilot.com/wiki/common-camera-shutter-triggering-for-sony-multiport-connectors-using-skysight-mono/
http://copter.ardupilot.com/wiki/common-pixhawk-auto-camera-trigger-without-chdk/

hamishwillee, 2016-01-26:
Absolutely. Done. (Stratosnapper was there, but not the others).

This  page has to be manually kept up to date. Do you think we could reasonably split into separate camera and camera gimbal landing pages? That way I could use our shortcode to dynamically generate the list of gimbals/cameras - one less thing to think about.

rmackay9, 2016-01-27:
@hamishwillee, yes, spliting cameras and gimbals would be ok I think.

As a side note, I'd like to create a "live video" section but I'm still working on content so just an FYI.

hamishwillee, 2016-01-27:
I will think about splitting this a bit more. In the meantime I extended the navmenu shortcode so that it can generate lists for a named page in the menu. This is now used instead of manual listing.

Closing as done :-)

PS Live video section makes sense, and that perhaps is the point we split into "Camera Triggering", Camera Gimbal, Camera Video or similar.

closed 2016-01-27 by hamishwillee