ArduPilot/ardupilot_wiki #279: Create SBUS Servo Output Info

R-Lefebvre, 2016-02-02:
Ardupilot/Pixhawk can be used with Futaba SBUS servos easily.  Need to create in info page.

rmackay9, 2016-02-02:
I think all that's required is that the user sets BRD_SBUS_OUT to 1.
@R-Lefebvre, anything else required?

hamishwillee, 2016-02-02:
We'd probably want to add a section to this article: http://planner.ardupilot.com/wiki/common-servo/#connecting_the_servo_to_a_pixhawk

I'm guessing you just plug the servo to the SBUS out pin (as [shown in this diagram](http://planner.ardupilot.com/wiki/common-pixhawk-overview/#pixhawk_pwm_connectors_for_servos_and_escs_and_ppm-sum_in_and_sbus_out)) and then set BRD_SBUS_OUT to 1?

Would also (I guess) be nice to list/link the servos that are known to work.

R-Lefebvre, 2016-02-24:
Yep, that is pretty much it.  Set BRD_SBUS_OUT to 1.  

You also need to set the SBUS address in each servo, but I don't think we need to explain that, other than to say it's required.  They would follow the manufacturer instructions for that.

The only servos I'm aware that would work, would be Futaba SBUS servos. I don't think anybody else is making SBUS compatible servos?  Not that I'm aware of anyway.

auturgy, 2016-03-19:
FrSky also make SBus servos, but Tridge's  recent addition, using SBus Out for additional PWM via a downstream converter, is probably worthy of a mention in the wiki too.

ivan95603, 2017-01-16:
Dear @R-Lefebvre,
Is it possible to use sbus servos without sbus receiver just over telemetry and PC joystick from mission planner?

Thanks

auturgy, 2017-01-16:
Yes, possible. Just use the sbus output on the Pixhawk. You need an sbus programmer to make sure your servos are numbered correctly.
Preferred to have rc though, if only as a backup.

Regards,

James

hamishwillee, 2017-01-30:
@auturgy @R-Lefebvre Can you guys make the changes so that this can be closed? If you can't/really don't want to let me know, and I'll take a shot at it.

auturgy, 2017-01-30:
I'm moving house, but can do it in a week or so.

Regards,

James

hamishwillee, 2017-01-30:
Thanks James, that would be awesome!

chandra2sekhar2000, 2017-04-18:
Hello,
i need to have SBUS output on the Pixhawk. i have set the BRD_SBUS_OUT = 300Hz on mission planner. i see that pulses start to come out of the pin, but the data coming from it is all zero's..i have connected sbus to serial print setup on arduino ..when i connect the arduino to sbus from x8R ,i can see the channel values varying from 200 to 1800..so arduino code is fine. but when i connect the sbus output from pixhawk to arudino i see that channels data is 0 ..
please kindly  guide me on setting up the SBUS out. all i need is to get the channels data from pixhawk via sbus out.Thank u

auturgy, 2017-04-18:
If you're using Copter you might just need to arm.

If using Plane:
Double check BRD_SBUS_OUT=n is correct

Then set RSSI_ANA_PIN=103 to make sure it isn't trying to read rssi from that pin.
You might need safety switch off (brd_safetyenable=0) for easier bench testing.

Then reboot and see if sbus is working.
Let me know how you go.

Regards,

James

auturgy, 2017-07-10:
@hamishwillee apologies for the length of time, but I went to look at this and your link is 404'd: http://ardupilot.org/planner/docs/common-servo.html#connecting_the_servo_to_a_pixhawk
Can you point me to where it sits in the wiki now?

OXINARF, 2017-07-10:
@auturgy http://ardupilot.org/copter/docs/common-servo.html#connecting-the-servo-to-a-pixhawk  That page isn't available in the MP wiki 🙂

brunoolivieri, 2019-08-02:
Looks like the PR #1039 add some information. It would be enough to close this issue?

rmackay9, 2019-08-03:
Yes, and I reworked it after this I think.  [The page is OK now](http://ardupilot.org/plane/docs/common-sbus-out.html) so we can close.

closed 2019-08-03 by rmackay9