ArduPilot/ardupilot_wiki #282: Problem when building ArduPilot for Pixhawk/PX4 on Linux with Make

joseluisUEX, 2016-02-05:
I'm trying to build ardupilot in Ubuntu 64, but when I make the code (make px4-v2) the following problem is returned:

CXX:     /home/pepe/ardupilot/libraries/AP_RSSI/AP_RSSI.cpp
/home/pepe/ardupilot/libraries/AP_RSSI/AP_RSSI.cpp:18:19: fatal error: utility: No such file or directory
 #include <utility>
                   ^
compilation terminated.
/home/pepe/ardupilot/modules/PX4Firmware/makefiles/module.mk:210: recipe for target 'libraries/AP_RSSI/AP_RSSI.cpp.o' failed
make[3]: *** [libraries/AP_RSSI/AP_RSSI.cpp.o] Error 1
make[3]: Leaving directory '/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/home/pepe/ardupilot'
/home/pepe/ardupilot/modules/PX4Firmware/makefiles/firmware.mk:231: recipe for target '/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build//home/pepe/ardupilot/module.pre.o' failed
make[2]: *** [/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build//home/pepe/ardupilot/module.pre.o] Error 2
make[2]: Leaving directory '/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
/home/pepe/ardupilot/modules/PX4Firmware/Makefile.make:143: recipe for target '/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4' failed
make[1]: *** [/home/pepe/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory '/home/pepe/ardupilot'
../mk/px4_targets.mk:96: recipe for target 'px4-v2' failed
make: *** [px4-v2] Error 2

Do you know how to solve the problem?

Jose Luis

hamishwillee, 2016-02-08:
Have you followed the instructions here: http://dev.ardupilot.com/wiki/building-px4-for-linux-with-make/#build ?

I just added this line to those instructions, and they now work.
```
git submodule init
git submodule update
```

peterbarker, 2016-11-15:
Closing; a lot has changed in the build systems since this issue was raised!

@joseluisUEX Please reopen this issue if you still are having troubles!

closed 2016-11-15 by peterbarker