ArduPilot/ardupilot_wiki #285: common-powermodule-landingpage/

lvale, 2016-02-05:
Hi Hamish

I believe that the power modules from http://www.mauch-electronic.com do deserve a specific page.

lvale, 2016-02-05:
Another one that deserves a mention 

http://arsovtech.com/?page_id=1546

hamishwillee, 2016-02-08:
Hi Luis

The problem is that I don't have these to test. Have you actually set one up?

Looking at http://diydrones.com/forum/topics/safety-first-100a-200a-hall-sensor-mit-bec-for-pixhawk-apm the connection is pretty straightforward (they have a diagram) but there is no mission planner setup information. Is the setup **exactly the same** as on [the 3DR Power module](http://copter.ardupilot.com/wiki/common-3dr-power-module/) so I could copy-paste? Ie can we select 3DPower module for settings. If not, then I'd need information on that.

The arsovtech one has even less information and I wouldn't attempt that. I could list it as an option on the top level common page though.

Or is we (you?) have contacts we could encourage them to generate the docs?

Regards
Hamish

lvale, 2016-02-08:
Hi Hamish

I've set up the ones from Mauch and is exactly as described on that link, with the added bonus that we receive a test report of the module and input the multipliers on the dialog box on MP.

I'll call the guys at arsovtech.

CraigElder, 2016-02-08:
Both should be added please.
ArsovTech is Philip Kocmoud and he is already a wiki editor.

lvale, 2016-02-08:
I've already asked Philip :)

hamishwillee, 2016-02-08:
Thanks. I'm looking into this. Does anyone known about the Battery Monitor 2 screen? Specifically it looks like you can install a second battery monitor to an analog pin, smbus or Bebop? Anything else much can be said about this?

[image]

hamishwillee, 2016-02-09:
I have split out [Power Module Configuration in Mission Planner](http://planner.ardupilot.com/wiki/common-power-module-configuration-in-mission-planner/) from [3DR Power Module](http://planner.ardupilot.com/wiki/common-3dr-power-module/) to reduce duplication required in these new pages.

I've then created the [Mauch page here](http://planner.ardupilot.com/wiki/common-mauch-power-modules/). Would appreciate your review!

@lvale Can you point Philip to the pages as an example of how he might approach them.

All OK?

lvale, 2016-02-10:
@hamishwillee 
Couldn't this pic also be added to the landing page, alongside the others ?

[image]

hamishwillee, 2016-02-10:
Hi @lvale 

Sure, I've added it to the [landing page](http://planner.ardupilot.com/wiki/common-powermodule-landingpage/) and into the [overview](http://planner.ardupilot.com/wiki/common-mauch-power-modules/#overview).

On the landing page I haven't gone to the effort of making it look consistent, but it is there :-)

Closing this now.

closed 2016-02-10 by hamishwillee

lvale, 2016-02-11:
@hamishwillee Something strange happened to the pic because I removed the green background and it returned :)

hamishwillee, 2016-02-11:
Argg. You are so demanding :-) Fixed!

lvale, 2016-02-11:
ROTFL :)