ArduPilot/ardupilot_wiki #286: PM (performance monitoring) Information

bionicbone, 2016-02-09:
[common-Downloading and Analyzing Data Logs in Mission Planner](http://copter.ardupilot.com/wiki/common-downloading-and-analyzing-data-logs-in-mission-planner/) - 

I think the detail on this page for PM (performance monitoring) Information is out of date for the Pixhawk.

In particular when running APM hardware NLoop was always 1000 as it detailed on the page, however for Pix it seems that the figure is now a constant 4000? 

Would it be possible to review the details in this section for Pixhawk, in particular NLon, NLoop, and MaxT. What should they be and at what levels should further investigation be required.

Regards
Kev

hamishwillee, 2016-02-09:
@bionicbone Thanks Kevin! I spoke to @peterbarker on this. For this specific case he says that NLoop has been 4000 for a long time on Pixhawk (perhaps since Copter has worked at 400Hz). 

Having a number here is deceptive as som people will still be running Copter on older hardware.  Additionally, Plane's gotten the ability to run at different clocks recently.

So proposal is to remove the absolute numbers in this doc, but leave the recommendations in terms of percentages. Seem reasonable to you?

bionicbone, 2016-02-10:
@hamishwillee thanks for checking this out.  
I had not made the connection with NLoop and the clocking speed. Now of course it makes more sense. 
Happy to remove the absolute numbers, perhaps cover the relationship with the clock speed though? or maybe it was just me missing that very logical point lol 
It sounds like the percentage is the most important aspect though when people are trying to understand issues.
Thanks again
Kev

hamishwillee, 2016-02-10:
@bionicbone you're most welcome. I've updated the NLoop docs to say:

> The total number of loops since the last PM message was displayed. This allows you to calculate the percentage of slow running loops (which should never be higher than 15%). Note that the value will depend on the autopilot clock speed.

Also closing this. Reopen if you think there is more to say.

closed 2016-02-10 by hamishwillee