ArduPilot/ardupilot_wiki #287: common-Mauch Power Modules

lvale, 2016-02-10:
[common-Mauch Power Modules](http://planner.ardupilot.com/wiki/common-mauch-power-modules/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

Hi Hamish

a few bits for better precision

<original>
This page explains how to set up the Mauch the HS-100-V2 (100A) and HS-200-V2 (200A) Power Modules to measure battery voltage and current consumption.
</original>

<suggestion>
This page explains how to set up the MAUCH power modules HS-050-xx*(50A), HS-100-xx*(100A) and HS-200-xx*(200A) to measure battery voltage and current consumption.
* xx is the indicator for:
LV = Vow Voltage 2-6S, maximum 28V 
HV = High Voltage 4-14S, maximum 60V

[image]

</suggestion>

<original>
Overview
Mauch has a number of PMs, which are compatible with Pixhawk, APM, Pixhawk lite, AUAV-X2, DroTek und CUAV Pixhack. The sensors boards are capable of delivering 100A continuous current for HS-100-V2 and 200A for the HS-200-V2 (without any time limit). The maximum burst current is 1200A@25’C and 800A@85’C for 1 second.

Sensor Board:

Current sensor is a “true hall sensor” up to 200A (ACS758-200U) or 100A (ACS758-100U)
Ultra-low noise power supply (LP2985-4.0) for current sensor and offset shifting circuit.
Microchip MCP601 operational amplifier for offset shifting.
LiPo voltage measurement with 1% resistor divider in factor 9:1.
2x10cm / AWG12 cable soldered to current sensor as standard size (Up to AWG8 possible).
6 pol cable connection to Pixhawk / APM (both sides DF-13 connector).
1x 10cm and 1x 20cm / 6 pole cable in the box to select a different cable length if necessary.
18mm x 29mm x 11mm / 7g without cables and shrinking tube.
BEC:

Input 2-6S LiPo / max. 28V
Output 5.35V / 3A -> +/- 0.05V –> Ripple 10mV (0.2%) at 1.5A output current.
Input wrong polarity protection, as well as Panasonic FM 220uF/35V input capacitor to prevent burn out of BEC from voltage spikes.
4 pole cable to sensor board with DF-13 connector.
47mm x 18mm x 11mm / 8g with cables and shrinking tube.
</original>

<suggestion>
Overview
Mauch has a number of PMs, which are compatible with Pixhawk, APM, Pixhawk lite, AUAV-X2, DroTek und CUAV Pixhack. The sensors boards are capable of delivering 100A continuous current for HS-100-V2 and 200A for the HS-200-V2 (without any time limit). The maximum burst current is 1200A@25’C and 800A@85’C for 1 second.

Sensor Board:

Current sensor is a “true hall sensor” up to 200A (ACS758-200U) or 100A (ACS758-100U)
Ultra-low noise power supply (LP2985-4.0) for current sensor and offset shifting circuit.
Microchip MCP601 operational amplifier for offset shifting.
LiPo voltage measurement with 1% resistor divider in factor 9:1.
2x10cm / AWG12 cable soldered to current sensor as standard size (Up to AWG8 possible).
6 pol cable connection to Pixhawk / APM (both sides DF-13 connector).
1x 10cm and 1x 20cm / 6 pole cable in the box to select a different cable length if necessary.
18mm x 29mm x 11mm / 7g without cables and shrinking tube.
BEC:

Input 2-6S LiPo / max. 28V
Output 5.35V / 3A -> +/- 0.05V –> Ripple 10mV (0.2%) at 1.5A output current.
Input wrong polarity protection, as well as Panasonic FM 220uF/35V input capacitor to prevent burn out of BEC from voltage spikes.
4 pole cable to sensor board with DF-13 connector.
47mm x 18mm x 11mm / 8g with cables and shrinking tube.

</suggestion>

<original>
The Mauch PM’s are not available in the Sensor selection list (at time of writing) so you will need to select Other and perform a manual calibration. Because the PM uses hall sensors you can ignore the recommendation to calibrate the sensor at >3Amps.

</original>

<suggestion>
Each sensor board comes with an final test result, which indicates the calibration values for voltage and current measurement. 

• On the mission planner’s INITIAL SETUP | Optional Hardware | Battery Monitor screen set the “Sensor” to “Other”.
• Enter the voltage divider from the final test result and press "TAB" or click out of the field. Then the "calculated battery voltage" should be within a few millivolt of the actual battery voltage.
• Enter the "Amperes per volt" from the final test result (A/V) and press "TAB" or click out of the field.

[image]
</suggestion>

hamishwillee, 2016-02-10:
Hi @lvale 

Thanks very much. I've most of the changes (though I put the image under the overview, and I couldn't find your changes in the sensor board etc section). Please check it out.

The best way to make these sorts of changes is to use the normal process - you do them yourself on the wiki and then I review. Do you have access?

Cheers
H

lvale, 2016-02-11:
no access :)

hamishwillee, 2016-02-11:
@lvale I'm going to give you access - can you email me: hamish at 3drobotics dot com with the email address you'd like associated with your login.

hamishwillee, 2016-02-11:
YOu are added now. Closing this :-)

closed 2016-02-11 by hamishwillee