ArduPilot/ardupilot_wiki #288: add Pixracer to common-autopilot's page

rmackay9, 2016-02-10:
http://copter.ardupilot.com/wiki/common-autopilots/

Most of the information can be found on these pages I think:
https://pixhawk.org/modules/pixracer
http://auav.com/

hamishwillee, 2016-02-11:
@rmackay9  OK, I have created the [draft here](http://planner.ardupilot.com/wiki/common-pixracer-overview/). It copies a lot from the pages you link.

Please let me know whether it is OK, or provide "enhancement suggestions".

Note that it is marked as under construction. That can go when you're happy with it.

hamishwillee, 2016-02-11:
Also, what should we say in http://planner.ardupilot.com/wiki/common-autopilots/common-choosing-a-flight-controller/ ?

Is it worth talking about this board yet? How would it fit into our "scale of recommendations"?

rmackay9, 2016-02-11:
Thanks very much Hamish!
I've made a minor change to highlight that it has only 6 pwm outputs.  That's the key limitation that allows it to be so small.
On the common-choosing-a-flight-controller page I think we should recommend it for small frames (i.e. 250 racing quads) that require no more than 6 pwm outputs.

hamishwillee, 2016-02-11:
Good change. I've removed the under construction warning, so this job is now done.

I added the recommendation you suggested to common-choosing-a-flight-controller - thanks!

Cheerio. Closing. H

closed 2016-02-11 by hamishwillee