ArduPilot/ardupilot_wiki #289: Add information about Mission Planner Battery Monitor 2 screen

hamishwillee, 2016-02-10:
Does anyone known about the Battery Monitor 2 screen? Specifically it looks like you can install a second battery monitor to an analog pin, smbus or Bebop? Anything else much can be said about this?

[image]

CraigElder, 2016-02-18:
This feature is very clear with the Pixhawk 2 that has two power module
ports attached.  It is backwards compatible option on a Pixhawk where some
of the pins are re-purposed from the way they are labelled.

hamishwillee, 2016-02-19:
So what pins do you connect on Pixhawk? The options in MP conveniently set this up for you by selecting pixhawk but we don't say anywhere what is setup :
[image]

In addition, the [BATT2_VOLT_PIN](http://copter.ardupilot.com/wiki/arducopter-parameters/#battery_voltage_sensing_pin_batt2_volt_pin) and [BATT2_CURR_PIN](http://copter.ardupilot.com/wiki/arducopter-parameters/#battery_current_sensing_pin_batt2_curr_pin) have same instructions as for the BATT versions: "On the Pixhawk powered from the PM connector it should be set to 3."
Obviously "3" is the value for the PM connector - what is the value for other connectors on pixhawk (and which once can we/should we use.

triolus, 2016-08-12:
Used pin A13 (adc 3.3v pin 4). It shows up on the 2nd battery monitor with a second brick. If someone can make A14 show up for current monitor instead of pin A12, which is the adc 6.6v, it would be fine...:). Of course you would have to fix the ratio. I would not bother with a12 adc 6.6v pin. It would spare one less wire routed across the board. BTW tested voltage with a power module from mauch, works great.:)

chopperxs, 2016-08-24:
I have just wired up my APM2.6 to a 3DR Power module for Batt and have used an Atto 180 to A1 for voltage of Batt2 and it does not work. I am only able to select the sensor type for Batt not Batt2 and even though I am getting a wrong result for Batt2 it does not display it on the display in MP (I added it to the screen but it always says Batt20 even though I have data streaming into the Initial Setup battery 2 section). I am using APM Plane. Has anyone out there actually got this to work properly?

CraigElder, 2016-08-30:
Please post user support requests to http://discuss.ardupilot.org/

closed 2016-08-30 by CraigElder

wildtypitch, 2017-12-13:
> Used pin A13 (adc 3.3v pin 4). It shows up on the 2nd battery monitor with a second brick. If someone can make A14 show up for current monitor instead of pin A12, which is the adc 6.6v, it would be fine...:). Of course you would have to fix the ratio. I would not bother with the A12 adc 6.6v pin. It would spare one less wire routed across the board. BTW tested voltage with a power module from mauch, works great.:)

@triolus How did you manage to get the batt 2 monitor to pick up A14 for current instead of A12. Bit ridiculous that this is still a problem. We can't be the only people monitoring two batteries?