ArduPilot/ardupilot_wiki #291: Copter Parameters - FS_BATT_ENABLE, FS_BATT_VOLTAGE, FS_BATT_MAH

bionicbone, 2016-02-12:
[Copter Parameters](http://copter.ardupilot.com/wiki/arducopter-parameters/) 

Hi,

I'm not sure part 1.16 of the documentation is clear.

FS_BATT_ENABLE describes the action for battery failsafe, 0=disabled, 1=Lane, 2=RTL.

Then we move onto FS_BATT_VOLTAGE and FS_BATT_MAH, which both suggest that if we fall below these values we "WILL" RTL. Does this "actually" depend on the value of  FS_BATT_ENABLE? i.e. the action would be RTL, LAND or if FS_BATT_ENABLE = 0 would be totally ignored?

I'd appreciate clarification as I have an issue raised against the APM Log File Analyser by a user because based on FS_BATT_ENABLE = 0 I'm stating the FS is not activated.

https://github.com/diydrones/APM_Log_File_Analyzer/issues/233

Thanks
Kev

bionicbone, 2016-02-13:
Based on this documentation:
http://copter.ardupilot.com/wiki/failsafe-battery/

I'd change the test for FS_BATT_ENABLE to:
Controls whether failsafe will be invoked when battery voltage or current runs low based on the values of parameters FS_BATT_VOLTAGE and FS_BATT_MAH.

I'd change the text for FS_BATT_VOLTAGE to:
Battery voltage to trigger failsafe. Set to 0 to disable battery voltage failsafe. If the battery voltage drops below this voltage then the copter will send a “Low Battery!” message via MavLink to the ground stations HUD (if using telemetry), activate a Loud beeping if a buzzer is attached, flash LEDs and based on the value of FS_BATT_ENABLE either Land, RTL or continue.

I'd change the text for FS_BATT_MAH to:
Battery capacity remaining to trigger failsafe. Set to 0 to disable battery remaining failsafe. If the battery voltage drops below this level then the copter will send a “Low Battery!” message via MavLink to the ground stations HUD (if using telemetry), activate a Loud beeping if a buzzer is attached, flash LEDs and based on the value of FS_BATT_ENABLE either Land, RTL or continue.

I hope this helps with a little copy and pasting.
Kev

hamishwillee, 2016-02-14:
@bionicbone Thanks for this. So yes, my interpretation of the docs is the same - FS_BATT_ENABLE controls whether any battery failsafe is enabled. FS_BATT_VOLTAGE and FS_BATT_MAH can be used to define what value the voltage or capacity failsafes will trigger at (assuming FS_BATT_ENABLE  is enabled) or whether the specific failsafe is enabled at all.

The documentation for copter parameters is managed in source and so tracked in the ardupilot repo. I've created a new issue for this here: https://github.com/diydrones/ardupilot/issues/3620 

Thanks very much.

closed 2016-02-14 by hamishwillee