ArduPilot/ardupilot_wiki #292: make sitl -j4 failed

MaartenHeuv, 2016-02-12:
Hello,

I don't succeed to get SILT running on a Windows 10 PC.
I started the whole process from the beginning 3x, but the first errors start to occur during the make of silt:

Maarten@Maarten-PC ~/ardupilot/ArduCopter
$ make sitl -j4
make: *** No rule to make target '/home/Maarten/ardupilot/modules/mavlink/message_definitions/v1.0/ardupilotmega.xml', needed by '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h'.  Stop.
make: *** Waiting for unfinished jobs....
Checking modules
modules/PX4Firmware/.git missing - need module init
+ git submodule init
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="bac29be0"
Submodule 'modules/PX4Firmware' (git://github.com/diydrones/PX4Firmware.git) registered for path 'modules/PX4Firmware'
Submodule 'modules/PX4NuttX' (git://github.com/diydrones/PX4NuttX.git) registered for path 'modules/PX4NuttX'
Submodule 'modules/gbenchmark' (git://github.com/google/benchmark.git) registered for path 'modules/gbenchmark'
Submodule 'gtest' (git://github.com/diydrones/googletest) registered for path 'modules/gtest'
Submodule 'modules/mavlink' (git://github.com/diydrones/mavlink) registered for path 'modules/mavlink'
Submodule 'modules/uavcan' (git://github.com/diydrones/uavcan.git) registered for path 'modules/uavcan'
Submodule 'modules/waf' (git://github.com/diydrones/waf.git) registered for path 'modules/waf'
+ git submodule update
Cloning into 'modules/PX4Firmware'...
remote: Counting objects: 140377, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 140377 (delta 29), reused 28 (delta 28), pack-reused 140335
Receiving objects: 100% (140377/140377), 69.83 MiB | 639.00 KiB/s, done.
Resolving deltas: 100% (103775/103775), done.
Checking connectivity... done.
Submodule path 'modules/PX4Firmware': checked out '3dcdec1fe106d00d7ab584b45db5e37b93ef0a22'
Cloning into 'modules/PX4NuttX'...
remote: Counting objects: 125539, done.
remote: Total 125539 (delta 0), reused 0 (delta 0), pack-reused 125539
Receiving objects: 100% (125539/125539), 31.16 MiB | 647.00 KiB/s, done.
Resolving deltas: 100% (102005/102005), done.
Checking connectivity... done.
Submodule path 'modules/PX4NuttX': checked out 'd48fa3072b2396c489966c4ab10183ac7cf3dea9'
Cloning into 'modules/gbenchmark'...
remote: Counting objects: 1788, done.
remote: Total 1788 (delta 0), reused 0 (delta 0), pack-reused 1788
Receiving objects: 100% (1788/1788), 454.06 KiB | 597.00 KiB/s, done.
Resolving deltas: 100% (1080/1080), done.
Checking connectivity... done.
Submodule path 'modules/gbenchmark': checked out '006d23ccca1375a973b7fae0cc351cedb41b812a'
Cloning into 'modules/gtest'...
remote: Counting objects: 7141, done.
remote: Total 7141 (delta 0), reused 0 (delta 0), pack-reused 7141
Receiving objects: 100% (7141/7141), 2.49 MiB | 638.00 KiB/s, done.
Resolving deltas: 100% (5321/5321), done.
Checking connectivity... done.
Submodule path 'modules/gtest': checked out 'c99458533a9b4c743ed51537e25989ea55944908'
Cloning into 'modules/mavlink'...
remote: Counting objects: 12862, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 12862 (delta 2), reused 0 (delta 0), pack-reused 12856
Receiving objects: 100% (12862/12862), 5.87 MiB | 667.00 KiB/s, done.
Resolving deltas: 100% (8168/8168), done.
Checking connectivity... done.
Submodule path 'modules/mavlink': checked out 'f8a672f6a08a9b06cacb8b9bacddba2590276904'
Cloning into 'modules/uavcan'...
remote: Counting objects: 16260, done.
remote: Total 16260 (delta 0), reused 0 (delta 0), pack-reused 16260
Receiving objects: 100% (16260/16260), 2.62 MiB | 599.00 KiB/s, done.
Resolving deltas: 100% (11012/11012), done.
Checking connectivity... done.
Submodule path 'modules/uavcan': checked out '6dd432c9742c22e1dd1638c7f91cf937e4bdb2f1'
Cloning into 'modules/waf'...
remote: Counting objects: 10119, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 10119 (delta 0), reused 0 (delta 0), pack-reused 10115
Receiving objects: 100% (10119/10119), 2.89 MiB | 597.00 KiB/s, done.
Resolving deltas: 100% (6856/6856), done.
Checking connectivity... done.
Submodule path 'modules/waf': checked out '878ab1e29494e28236a260212a5021b1de35d9a3'
+ cat
==============================
git submodules are initialised

Please see http://dev.ardupilot.com/wiki/git-submodules/

Please restart the build
==============================
+ exit 1
D:/cygwin/home/Maarten/ardupilot/mk/modules.mk:8: recipe for target 'CHECK_MODULES' failed
make: *** [CHECK_MODULES] Error 1

Please help me, as I'm not very experienced in the Linux (Cygwin) environment

Thanks in advance Maarten

hamishwillee, 2016-02-14:
@MaartenHeuv This doesn't work for me either. It hasn't worked for several weeks and I have been unable to debug. @tridge - your advice/help would be appreciated.

hamishwillee, 2016-02-14:
@MaartenHeuv Depending on what you want to do with SITL, there may be other options - e.g. DroneKit-SITL allows you to do simulation without having to build the platform. Useful if you just want something to test MAVLink messages against and you don't need the latest and greatest platform. 

It is also possible to build on Windows in Vagrant - http://dev.ardupilot.com/wiki/setting-up-sitl-using-vagrant/

rmackay9, 2016-02-15:
This must be related to the move of the mavlink repo to a submodule.  I would think that "git submodule init", "git submodule update" would fix the problem but I haven't actually tried it.
Although I'm a windows user, I personally always run SITL in a virtual machine.  http://dev.ardupilot.com/wiki/setting-up-sitl-on-windows/
I know that's not likely the answer you'd like, that's how I know it works though.

hamishwillee, 2016-02-15:
@rmackay9 Do you just run that in the root of ardupilot? I ask, because I've already run those commands, to no joy. My log looks like:
```
$ make sitl -j4
echo "Generating MAVLink headers..."
Checking modules
Generating MAVLink headers...
#goto mavlink module directory and run the most recent generator script
echo "Generating C code using mavgen.py located at" /home/hamis/ardupilot/modules/mavlink/
Generating C code using mavgen.py located at /home/hamis/ardupilot/modules/mavlink/
PYTHONPATH=/home/hamis/ardupilot/modules/mavlink/ python /home/hamis/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py --lang=C --wire-protocol=1.0 --output=/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0 /home/hamis/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="92fea608"
Validating /home/hamis/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Parsing /home/hamis/ardupilot/modules/mavlink//message_definitions/v1.0/ardupilotmega.xml
Note: message DATA64 is longer than 64 bytes long (74 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message DATA96 is longer than 64 bytes long (106 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message REMOTE_LOG_DATA_BLOCK is longer than 64 bytes long (214 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Validating /home/hamis/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Parsing /home/hamis/ardupilot/modules/mavlink//message_definitions/v1.0/common.xml
Note: message GPS_STATUS is longer than 64 bytes long (109 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ATTITUDE_QUATERNION_COV is longer than 64 bytes long (76 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GLOBAL_POSITION_INT_COV is longer than 64 bytes long (193 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOCAL_POSITION_NED_COV is longer than 64 bytes long (237 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIGHRES_IMU is longer than 64 bytes long (70 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_SENSOR is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SIM_STATE is longer than 64 bytes long (92 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FILE_TRANSFER_PROTOCOL is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 176 MAVLink message types in 2 XML files
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega
Generating C implementation in directory /tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/common
Copying fixed headers
C:/cygwin/home/hamis/ardupilot/mk/mavgen.mk:9: recipe for target '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h' failed
make: *** [/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h] Aborted (core dumped)
make: *** Deleting file '/tmp/ArduCopter.build/libraries/GCS_MAVLink/include/mavlink/v1.0/ardupilotmega/mavlink.h'
```

rmackay9, 2016-02-15:
@hamishwillee, yes, in any directory including "ardupilot" or lower (like "ArduCopter").  The coredump doesn't look good!

hamishwillee, 2016-02-15:
:-( Any suggestions on how to fix, or do I need to chase Tridge?

MaartenHeuv, 2016-02-15:
@hamishwillee 
"Good" to hear that I'm not alone. I sometimes have doubts about myself.

I simply want to do plane simulations using my APM with MP simulation. I have not the intention to develop or debug APM software.
Did you try the run the simulator with help of the Vagrant tool in an VM environment ?

Best regards Maarten

hamishwillee, 2016-02-15:
I have in the past, and it should work because it is effectively in Linux.

magicrub, 2016-02-15:
I just broke my Ubuntu virtual machine image so I wouldn't mind trying SITL
out ASAP on Windows. I'll give it a try.

hamishwillee, 2016-02-15:
@magicrub That would be really helpful!

hamishwillee, 2016-02-16:
Hi @MaartenHeuv 

Right now you can get this to work (tested) by doing 
```
make sitl -i
sim_vehicle.sh -j4 --map
```
(instead of `make sitl -j4`)
In the near future master will be patched so that the wiki instructions work.

This solution is discussed in https://github.com/diydrones/ardupilot/issues/3586. Thanks @magicrub for your help!

I am closing this. 

Regards
Hamish

closed 2016-02-16 by hamishwillee

magicrub, 2016-02-16:
> This solution is discussed in diydrones/ardupilot#3586

Fix is in master