ArduPilot/ardupilot_wiki #294: Advanced Compass Setup

justinsupreme, 2016-02-15:
[common-Advanced Compass Setup](http://copter.ardupilot.com/wiki/common-compass-setup-advanced/) 

I want to point out an issue that sent my Beautiful copter on a crash coarse.    If the there is no external compass on a Pixhawk, compass 1 is the internal compass.    If there is an external compass, it becomes compass 1 and the internal compass becomes compass 2...or so it seems.   

I thought that compass 1 was always the internal compass so I unchecked the externally mounted box for compass 1, said use compass two as primary compass and clicked the externally mounted box for compass 2.

I took out for a test flight and it went out of control and F***ed my Lumenier QAV540 v2 all up.   Lots of money and a disappointing day.

I believe compass 1 should always be the internal compass to avoid confusion.    Had this been clear I could have saved hundreds of dollars.   I love mission planner and my pixhawk, but I see a lot of issues that could be cleared up with more clear direction.

I still don't fully understand why it sent my copter crazy, but it did.

hamishwillee, 2016-02-15:
Hi @justinsupreme 

The logic behind the compass is that *we always recommend that you use compass 1, and that this will be the external compass if available*. I am sorry that this was not the logic that you used, but it is [very clearly documented in the overview here](http://copter.ardupilot.com/wiki/common-compass-setup-advanced/#overview). Arguably this is the best approach, since compass 1 is what gets selected by default.

If you feel that this is not reasonable, then please post an issue in the ardupilot repo (this repo is to document "what is", not "what we would like").

I'm very sorry that your copter crashed. That is never a nice feeling.

closed 2016-02-15 by hamishwillee

justinsupreme, 2016-02-15:
Why would it allow you to set compass 2 as the primary.   I found this wiki
page last night before I commented.   It is not invalid because many have
crashed from the same thing.   I did my research.

I am not trying to say it was not documented.   I just feel like a little
more clarification right from mission planner would be nice so people don't
make the same mistake.  It is counter intuitive.

hamishwillee, 2016-02-15:
@justinsupreme The only way to prevent false assumptions is to read the manual. 

I completely agree that highlighting the external/internal compass in the mission planner would reduce the chance of this issue happening. I have created this issue for you to track: https://github.com/diydrones/MissionPlanner/issues/1141

justinsupreme, 2016-02-15:
> created this issue for you to track: diydrones/MissionPlanner#1141

I'm not trying to be contrary.   I am just trying to help.  The manuals
have gotten better.   Thanks for the response.

hamishwillee, 2016-02-16:
Thanks, I appreciate that.

hamishwillee, 2016-02-28:
@justinsupreme The mission planner will now have visual cue that which is the primary compass as per linked item above. This may take a little while to get out of beta.