ArduPilot/ardupilot_wiki #298: Commercial Support

virtualrobotix, 2016-02-26:
[Commercial Support](http://planner.ardupilot.com/wiki/common-commercial-support/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant
**
Laser Navigation SRL ( Virtualrobotix )**

Via Matteotti,34
Calcio (BG) - 24054 - 
Italy

Tel.: +390363968520
Email: info@virtualrobotix.com
website and webshop : www.virtualrobotix.it 
community : www.virtualrobotix.com 
skype account : virtualrobotix

Capabilities:

Autopilot hardware design : 

1. VR Brain 5 : Professional Level Flight Control Board  support : APM Copter , Plane , Rover based on STM32F4 micro controller.
2. VR uBrain 5 : Entry Level Flight Control Board  support : APM Copter , Plane , Rover based on STM32F4 micro controller.
3. VR Brain LX : R&D support : APM Copter, Plane , Rover based on RPI2 core CPU
4. VR Brain 6 core : Professional Level Flight Control Board  support : APM Copter , Plane , Rover based on STM32F4-F7 (WIP) micro controller.
5. VR GPS 8 : Advanced GNSS system support GPS , GLONASS , EGNOSS, BEIDU constellation and integrate HMC5983 magnetometer.
6. VR Link : Telemetry module on 868 - 433 mhz support VR Brain and uBrain .

Software and firmware development : 

- Support customization of Nuttx operating system and ArduPilot functionality.
- Support customization of Mission Planner and VR Pad Station application ( Ground Station based on Android Operating Systems).
- Vehicle design, integration and setup
- Troubleshooting and support (log review, vehicle support)
- New vehicle creation and support
- Custom design of advanced Companion Computer , integration with ROS and support in development of SLAM advanced functionality and advanced payload managment.

Advanced support of startup project : some kickstarter drone company develop its products with support of our dev team.

magicrub, 2016-02-28:
Who approves these? With the latest announcement this should go in ASAP.

hamishwillee, 2016-02-28:
@virtualrobotix Thanks Robert, this has been added to http://planner.ardupilot.com/wiki/common-commercial-support/#laser_navigation_srl_virtualrobotix and http://dev.ardupilot.com/wiki/common-commercial-support/ . It will automatically be copied to the copter, rover, and plane wikis in a few minutes.

@magicrub The criteria for acceptance are on the page:

> To be added to this page we require that the applicant has made a contribution to ArduPilot or a related project in some way. That includes code, documentation or helping people out on the forums. The intention with this policy is to ensure that companies listed here will be active participants in the community."

So basically if I know them I will add them, otherwise I will wait on someone like you or @CraigElder to confirm. 

I do urgent stuff if only I can reasonably do it, but generally most weekends I am unavailable. If something needs to be done quickly then feel free to do it yourself  and then me or Randy or Grant or Tridge can approve them :-) I don't want to be a bottleneck.

I believe this is closed.

closed 2016-02-28 by hamishwillee