ArduPilot/ardupilot_wiki #299: Adding a new MAVLink Message (Code Overview) has bugs

hamishwillee, 2016-02-28:
[Adding a new MAVLink Message (Code Overview)](http://dev.ardupilot.com/wiki/code-overview-adding-a-new-mavlink-message/) is out of date

1. What is the autogeneration feature?
1. What version does it come in?
1. The docs say

   >Step #3: Add the new message definition to the common.xml or ardupilotmega.xml file

   But these links are broken.
 
   > Step 4 says you need to call ./libraries/GCS_MAVLink/generate.sh

What is the actual story here? Is it a matter that if you update the master definition in https://github.com/diydrones/mavlink/tree/master/message_definitions and rebuild ardupilot then step 3 and 4 are done for you? In what versions is this true. 

1.Do we need to talk about the "old version

hamishwillee, 2016-02-28:
@rmackay9 @tridge Can you please tell me what the real story is here. We can have a skype for it if you want.

sediq-khan, 2019-02-07:
Hi,
I am also adding a custom message to ardupilotmega.xml, I can not do the step 4, since I can not find a file called "./libraries/GCS_MAVLink/generate.sh", do you have any tutorial where step 4,5 and 6 is done.

Thank you

brunoolivieri, 2019-08-02:
The links are not broken anymore.

@sediq-khan Have you tried the Gitter devel channel?

auturgy, 2019-08-03:
The current documentation is correct: mavlink is auto generated during waf compile, so that old step is no longer required. See http://ardupilot.org/dev/docs/code-overview-adding-a-new-mavlink-message.html
Closing, as this issue should have been closed long ago when the docs were updated.

closed 2019-08-03 by auturgy