ArduPilot/ardupilot_wiki #300: common-Accelerometer Calibration in Mission Planner

1Steven, 2016-02-29:
Unwisely, I upgraded to APM Plane 3.5 and MP 1.3.35 at the same time and now the accel calibration no longer works using the 3DR modem radio.  What's up?

magicrub, 2016-02-29:
It should work, what's the problem?

1Steven, 2016-02-29:
Hello, 
When I initiate accel calibration for Plane, it never starts and will eventually time out. This morning, I retrograded to the prior APM Plane firmware and it works again. 
I found another issue too. I use battery mAh used in the HUD. In the latest MP stable release, it does not record mAh used unless I go to User Items, uncheck it and then check it again. My MinimOSD sees the mAh used consistently, so I'm pretty sure it's on the MP side.
Thanks,
Steven

magicrub, 2016-02-29:
This is all related to a possible Mission Planner bug, please post an issue
there instead of the wiki.
https://github.com/diydrones/MissionPlanner/issues/new

-TomP

1Steven, 2016-02-29:
Done...

closed 2016-02-29 by 1Steven