ArduPilot/ardupilot_wiki #303: Communicating with Raspberry Pi via MAVLink

trondhindenes, 2016-03-06:
[Communicating with Raspberry Pi via MAVLink](http://dev.ardupilot.com/wiki/raspberry-pi-via-mavlink/) - Couple of comments on the "communicating with raspberry pi" section
1. I'm not sure the +5v connection from the pixhawk to the pi should be made. I'm running without it. Common ground yes, but the +5v is not needed, as the pi has its own power supply thru usb, and the GPIO +5 header is un-regulated, which introduces risk of damaging the pi.

2. the "screen -x" command should be "sudo screen -x". The reason for this is that any commands started inside of rc.local will run as root. Thereby, simply using screen -x will fail, as there are no screen instances running for the current user.

hamishwillee, 2016-03-06:
Hi @trondhindenes 

Thanks very much.

The image doesn't show the RPi being powered by USB - is it really the case that USB port is used to power the RaPi in the air? I'd be happy to remove the existing power line if we showed that/how the USB power was connected on a real vehicle. Thoughts?

 I've fixed "2".

Regards
H

trondhindenes, 2016-03-06:
I was just thinking in terms of the Rpi, the recommendation is to power it using the usb in, since that power in put is very regulated and fused to ensure nothing bad happens. As far as I understand, the GPIO power in doesn't have any of those failsafes. I'm using a separate 5V bec hooked up to the usb power on the rpi in my project, but maybe the documentation page should simply list that both the GPIO and USB are valid ways of powering the rpi, and if the last one is used, the +5V gpio pin should remain disconnected from the pixhawk.

Just trying to clarify for others as I go about learning :-)

hamishwillee, 2016-03-06:
Thanks very much for the help! I haven't tried this, so trying to make sure I don't break anything in the docs :-) 

What you say makes sense. I've added your clarification, which you can see here: http://dev.ardupilot.com/wiki/raspberry-pi-via-mavlink/#connecting_the_pixhawk_and_rpi

[image]

OK to close this?

trondhindenes, 2016-03-06:
Looks great, thanks! Closing.

closed 2016-03-06 by trondhindenes

tkoch96, 2017-12-01:
Hello, I have a question regarding this issue. I'm trying to power a raspberry pi 3 model B with a Pixhawk v2. 
The telemetry port does not supply sufficient power for the Pi to operate. The Pi will try to boot, but when the current draw approaches ~.5A (measured using an ammeter), the Pi will turn off and on again. The Pixhawk 2 documentation (http://www.hex.aero/wp-content/uploads/2016/07/DRS_Pixhawk-2-17th-march-2016.pdf)  says the telemetry port limits current to 1A and total peripheral device current to 2.5A so this kind of makes sense.

I'm not quite sure what is meant by 'powering via USB' (referencing the comment above). There is a pin labeled 'USB' on the Pixhawk through which the buzzer is connected, but this has a current limit of 250 mA (according to the documentation above) and only provides 3.6V or so. 

Does anyone have any recommendations for powering my Pi? I'm thinking of one of two ways:
1. Put a small battery pack on the drone with a voltage regulator solely for powering the Pi.
2. Somehow siphon power directly from the drone battery through a 5V regulator. Not sure exactly how this is done.

Can anyone provide insight?