ArduPilot/ardupilot_wiki #305: Copter Mission Command List - HoldTimeFromLaunch

gaucho1978, 2016-03-09:
[Copter Mission Command List](http://copter.ardupilot.com/wiki/mission-command-list/) 
the copter shall start the movement from a waypoint to the next after a programmed number of seconds elapsed from the launch of the mission, 
On DJI products (and on ardupilot) instead, the hold time is measured from the moment when the waypoint is reached, alter the stabilization phase.
Is this achievable with ardupilot?

magicrub, 2016-03-09:
See the work-in-progress of DO_CONDITIONAL_DELAY command (soon to be
renamed) work by Randy: https://www.youtube.com/watch?v=9VK3yjIyCSo

gaucho1978, 2016-03-09:
thank you very much. this feature seems to accept UTC time date as parameter. this means that you have to program the future time of execution of the mission, wait for that day, execute the mission, then you have to modify the mission file for a new flight. How is it supposed to determine the new UTC dates in the mission for a new flight? manually?
I was asking an easier function, where the time is calculated with reference to the mission start event.
WAIT time = elapsed seconds from launch - T

where can i read about the work in progress of this function?
It is possible that better thinking about it, this solution may work also for us.
I have a problem of syncronization of the copter with the rotation of an antenna to solve.
Thank you

magicrub, 2016-03-09:
you can omit info to treat it as a don't-care. For example, if the day, hour, and minute are 0 then the task will delay until only the seconds line up.

For your antenna rotation thing, have you tried the antenna Tracker (ArduTracker)?

gaucho1978, 2016-03-09:
ok. thank you. Hour 0 and minute 0 exists, so may be it is better to use -1 to set that value as "don't care". 
Thank you for the explanation.
thank you even for the antenna tracker. i know about it but it is not useful in my case.

magicrub, 2016-03-09:
Yes, using -1 has been discussed. The feature is still a work-in-progress
by @Randy.

gaucho1978, 2016-03-09:
if we can set also the "don't care values", this feature perfectly match my needs.

magicrub, 2016-03-09:
This is not a wiki problem so please close this wiki issue. Please visit the discussions here:
issue: https://github.com/diydrones/ardupilot/issues/3675
pull-request: https://github.com/diydrones/ardupilot/pull/3708

closed 2016-03-09 by gaucho1978