ArduPilot/ardupilot_wiki #308: Problem running SITL on Windows (with or without Mavproxy)

HuguesD, 2016-03-12:
I followed precisely instructions on (http://dev.ardupilot.com/wiki/sitl-native-on-windows/) to run Sitil on Windows (Win 7). The installation part goes as documented (Cygwin install, MavProxy install).
It fails though when trying to run it, and it fails in the two possible modes : with or without mavProxy.

1-With MavProxy I run the command line as indicated in the doc and it fails with the following output:

Hugues@Hugues-PC ~/ardupilot/Arducopter
$ sim_vehicle.sh -j4 --map
~/ardupilot/Arducopter ~/ardupilot/Arducopter
Building sitl
Checking modules
// BUILDROOT=/tmp/Arducopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="4bf3ec0e"
~/ardupilot/Arducopter
Starting up at CMAC : -35.363261,149.165230,584,353
Starting ardupilot : /home/Hugues/ardupilot/Arducopter/Arducopter.elf -S -I0 --home -35.363261,149.165230,584,353 --model --speedup=1
Window access not found, logging to /tmp/ardupilot.log
../Tools/autotest/run_in_terminal_window.sh: line 28:  6004 Segmentation fault      (core dumped) $cmd $* &> $filename < /dev/null

The Mavproxy console and map window are displayed. The MacProxy console indicates the following error message:
Exception in thread main_loop:
Traceback (most recent call last):
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
....
What is thus user "Stephen" ? Is this something hardcoded somewhere ?

2-Then I tried the second method without MavProxy and it fails too:

Hugues@Hugues-PC ~/ardupilot/Arducopter
$ ./ArduCopter.elf --home -35,149,584,270 --model quad
Started model quad at -35,149,584,270 at speed 1.0
Starting sketch 'Arducopter'
Starting SITL input
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
ERROR: Floating point exception - aborting
Aborted (core dumped)

Am I doing something wrong or is this known under windows and how to solve it ?

Thx

magicrub, 2016-03-12:
Is this on the latest master?

HuguesD, 2016-03-12:
Yes, I did a bit to compile the latest master

hamishwillee, 2016-03-14:
That will be @stephendade who maintains MAVProxy on windows. I've asked if he has any insight [on gitter](https://gitter.im/Dronecode/MAVProxy?at=56e609a40055f8f35a82d761).

stephendade, 2016-03-14:
Using the latest Ardupilot source and 1.4.42 of MAVProxy (uploaded last night), I can't replicate the issue on my computer. Arducopter and Arduplane run quite nicely.

@HuguesD - could you run a "make clean" on Ardupilot and try the latest version of MAVProxy? Let me know if it's still an issue.

HuguesD, 2016-03-14:
Ok it works now. Thx.
If I want to add a MavProxy output to a COM port (for testing a real antenna tracker that will connect via 3DR radio on SITIL), the documented command line OUTPUT ADD produces a syntax error. What is the correct syntax ?

hamishwillee, 2016-03-14:
@HuguesD Suggest you ask that on the mavproxy gitter channel and cross post if it turns out the output add is actually incorrect

closed 2016-03-14 by hamishwillee

HuguesD, 2016-03-15:
Thx Hamish, but what is gitter ?

hamishwillee, 2016-03-15:
Gitter is github's repository based chat room. I refer (and link to it) here: https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/308#issuecomment-196090606

priyamshah112, 2018-12-29:
SIM_VEHICLE: Using defaults from (/home/PRIYAM SHAH/ardupilot/Tools/autotest/default_params/sub.parm)
SIM_VEHICLE: Run ArduSub
SIM_VEHICLE: “/home/PRIYAM SHAH/ardupilot/Tools/autotest/run_in_terminal_window.sh” “ArduSub” “/home/PRIYAM SHAH/ardupilot/build/sitl/bin/ardusub” “-S” “-I0” “-” “–model” “vectored” “–speedup” “1” “–defaults” “/home/PRIYAM SHAH/ardupilot/Tools/autotest/default_params/sub.parm”
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: “mavproxy.exe” “–master” “tcp:127.0.0.1:5760” “–sitl” “127.0.0.1:5501” “–out” “127.0.0.1:14550” “–out” “127.0.0.1:14551” “–out” “udp:0.0.0.0:14550” “–map” “–console”
RiTW: Starting ArduSub : /home/PRIYAM SHAH/ardupilot/build/sitl/bin/ardusub -S - --model vectored --speedup 1 --defaults /home/PRIYAM SHAH/ardupilot/Tools/autotest/default_params/sub.parm
which: no osascript in (/usr/local/bin:/usr/bin:/cygdrive/c/sdk-tools-windows-3859397/platform-tool::/cygdrive/c/sdk-tools-windows-3859397/tools/bi::/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/Python 3.6:/cygdrive/c/Python 3.6/Scripts:/cygdrive/c/Program Files/Java/jdk1.8.0_131/bin:/cygdrive/c/Program Files/Java/jre1.8.0_131/bin:/cygdrive/c/TDM-GCC-64/bin:/cygdrive/c/Program Files/nodejs:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/TDM-GCC-64:/cygdrive/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit:/cygdrive/c/WINDOWS/IsmFiles:/cygdrive/c/MinGW/bi:/cygdrive/c/MinGW/msys/1.0/bin::/cygdrive/c/firefoxdrive:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/c/Program Files/Geth":/cygdrive/c/Program Files/Git LFS:/cygdrive/c/Program Files/PuTTY:/cygdrive/c/Program Files (x86)/MAVProxy:/cygdrive/c/Anaconda:/cygdrive/c/Anaconda/Library/mingw-w64/bin:/cygdrive/c/Anaconda/Library/usr/bin:/cygdrive/c/Anaconda/Library/bin:/cygdrive/c/Anaconda/Scripts:/cygdrive/c/Users/PRIYAM SHAH/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/PRIYAM SHAH/AppData/Roaming/npm:/cygdrive/c/Users/PRIYAM SHAH/AppData/Local/GitHubDesktop/bin:/cygdrive/c/Users/PRIYAM SHAH/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/phantomjs-2.1.1-windows/bin:/cygdrive/c/Users/PRIYAM SHAH/AppData/Local/Google/Cloud SDK/google-cloud-sdk/bin:/cygdrive/c/Program Files/heroku/bin:/home/PRIYAM SHAH/ardupilot/Tools/autotest)
RiTW: Window access not found, logging to /tmp/ArduSub.log
SIM_VEHICLE: Keyboard Interrupt received …
SIM_VEHICLE: Killing tasks

Occurrs And the Log written is

Set parameter SIM_SPEEDUP to 1.000000
Creating model vectored at speed 1.0
Home: 33.810314 -118.393869 alt=0.000000m hdg=270.000000
Starting sketch ‘ArduSub’
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection …
Failed to load defaults from /home/PRIYAM
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
TSYS01 device is null!

Can Someone help me out!
System : Windows 10

PrasadNR, 2019-10-21:
Hey @priyamshah112 , were you able to figure out a solution? I am getting the same error

> RiTW: Window access not found, logging to /tmp/ArduSub.log

Reference: https://github.com/ArduPilot/ardupilot/pull/12006#issuecomment-544586009