ArduPilot/ardupilot_wiki #310: GCS Failsafe

shening, 2016-03-19:
[GCS Failsafe](http://copter.ardupilot.com/wiki/gcs-failsafe/) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant

Hi,

I am using Arducopter 3.2 and I can not get the GCS failsafe to trigger at all.

magicrub, 2016-03-19:
I see you've posted this on the wiki issue board. Is this a feature failure
or a documentation problem?

rmackay9, 2016-03-19:
I think this is a vehicle issue and the problem is likely that @shening expects (perhaps rightly so) that the GCS failsafe will trigger whenever the vehicle loses contact with the ground station.  Sadly this is not how it works.  It only triggers when the user is using the ground station to manually control the vehicle using rc-overrides.  There's an issue on the to-do list to make the GCS failsafe work like the Plane failsafe .. which would make it trigger whenever it loses contact with the GCS.

closed 2016-03-19 by rmackay9