ArduPilot/ardupilot_wiki #326: Dangerous advice?

fnoop, 2016-03-28:
http://ardupilot.org/copter/docs/ac_tipsfornewpilots.html

There was a post recently on diydrones where a novice was trying to build a meter size (T960) hex with 16" CF props.  It was flipping initially and when esc order was sorted it was then leaning significantly when trying to take off.  This wiki page was cited by several people as advice to proceed and what they had done in the past.
```
" Hold the copter firmly in your hand and level in the air and with the props clear of your body, advance the throttle a bit to get a feel for what’s going on. "
```
Is this not potentially incredibly dangerous advice for newbies?  I wouldn't like to think of the potential harm that could come from large CF props if a copter tried to flip while throttle raised above someone's head.  Perhaps I'm just being silly/unreasonable, but thought I would raise it for discussion.

hamishwillee, 2016-03-30:
Thanks @fnoop 

While I'd hope readers might have a little common sense, I've put this suggestion inside a warning. You can see it at the link now.

rmackay9, 2016-03-31:
I agree that we could remove the advice about holding the copter.  I personally never do that.  I think it's not necessary especially now that we have a feature that allows the user to spin each motor individually to test it's direction.

hamishwillee, 2016-03-31:
@rmackay9 So where's the information on how to do that (individual spin motor) thing.

fnoop, 2016-03-31:
@hamishwillee here it is
http://ardupilot.org/copter/docs/connect-escs-and-motors.html#testing-motor-spin-directions

hamishwillee, 2016-04-10:
Thanks very much @fnoop @rmackay9 - I've fixed this up in #367

rmackay9, 2016-04-11:
nice, thanks.

closed 2016-04-11 by rmackay9