ArduPilot/ardupilot_wiki #331: Camera Gimbal with Servos

trondhindenes, 2016-03-28:
[Camera Gimbal with Servos](http://ardupilot.org/copter/docs/common-camera-gimbal.html)

I consider myself a pretty technical savvy person, but the documentation regarding Gimbal setup completely confuses me. Would this be reviewed/rewritten into something clearly understandable? 
I don't understand the difference between a servo-based vs the other types of gimbals (I mean its all pwm input right?) The MP screenshot seems to be from a very old version of the software and doesn't look like the gimbal screen on my software. I don't understand the mapping between input CH version tilt/roll/pan channel. There should be a complete end-to-end setup guide for configuring modern self-stabilizing gimbals. This is the only part of the PixHawk experience where I completely draw a blank, and not getting much help from the documentation.

gmorph, 2016-11-15:
We would love to get a PR with a newly updated page on this.  But at the moment most users seem to be fine with the current page.
Note the page you reference is for Servo Gimbal's only.  Not Gimbal's with motors.
The mapping it mapping your input channel to your output channel.  So you transmitter may use RC6 for input for you to control the gimbal but you can map that to RC9 on the output of the Pixhawk.  So your RC6 control inputs get sent to the RC9 output channel on the Pixhawk.
Hope that helps.
Thanks, Grant.

hamishwillee, 2016-11-16:
Hi @gmorph 
I'm reopening this as I encountered similar problems. We really need an explanation of the types of gimbals, and if our information only applies to a particular type of gimbal then we should make that very clear at the beginning of the document.

gmorph, 2016-11-17:
Ok - I'll have a look and see if I can make it clearer.

auturgy, 2017-10-15:
@gmorph bump.

brunoolivieri, 2019-08-02:
Hi, 

This suggestion still makes sense? The documentation still the same?

Should we close the issue or leaves it as a desirable?

rmackay9, 2019-08-03:
We should probably update the images on the servo gimbal page.  It's not terrible though..

This user really wanted to set up a brushless gimbal and so certainly the page would have been confusing.   He should have looked at one of the other three pages we have (the SToRM32, the SimpleBGC or the Tarot setup page).  By the way the [SimpleBGC page](http://ardupilot.org/copter/docs/common-simplebgc-gimbal.html) has a significant number of formatting issues.

brunoolivieri, 2019-08-06:
Indeed the images could be updated. If it is important, I can fix it later.

Regarding the [SimpleBGC page](http://ardupilot.org/copter/docs/common-simplebgc-gimbal.html), I did the PR #1880 to solve some of the issues of the page.