ArduPilot/ardupilot_wiki #333: Onboard Mag Calibartion

Jman841, 2016-03-28:
Currently there is no documentation on how to use the ON board mag calibration for Arduplane. I want to test this out as we are about to build a large plane that will be difficult to do the traditional method with. However, I am not sure how to do this. Do we take off and go to RTL or guided mode and start this? 

Thank you

rmackay9, 2016-03-31:
There may be two versions of the on-board compass calibration now.  I'm not sure which Plane uses so I think it's @tridge or @magicrub or @gmorph who would know best.
So there's a new on-board compass calibration that, from the user's point of view, is just like the existing "Live Calibration" in that the vehicle is disarmed and the user waves it around.
The other compass calibration that's done on-board is better known as "compass learn" in which the compass offsets are learned in flight.  previously there was a COMPASS_LEARN parameter that should be set to "1" to enable this (although I think on plane it's enabled by default).

Basically my guess is that there's no need for @Jman841 to do anything special except maybe fly the vehicle around a bit.  I think most people do this in a more manual flight mode.

auturgy, 2017-10-15:
@jschall would you be able to link your video tutorial in here: http://ardupilot.org/copter/docs/common-compass-calibration-in-mission-planner.html with a few notes, so we can close this?  Perhaps replace Randy's video, as it's a little dated.

brunoolivieri, 2019-08-02:
Looks like there are newer documentation that covers the Issue.

Could we close it?

rmackay9, 2019-08-03:
Yes, we've updated the compass calibration page since this issue was raised so I think it can be closed now.  Jonathan's video has been linked at the bottom of the page.

closed 2019-08-03 by rmackay9