ArduPilot/ardupilot_wiki #337: Create epub builds by default of the wikis

hamishwillee, 2016-03-28:


jpkh, 2016-03-29:
Do we want this to be build on every update or once day/week?

hamishwillee, 2016-03-29:
I think once a week or so.

jpkh, 2016-04-08:
just ran some tests for this and 2 Android epub softwares seems to crash on sphinx created files. Have you tested and readers?

hamishwillee, 2016-04-08:
I haven't tried this epub build at all - not high on my priority list. I have previously run the epub output on a kindle though.

jpkh, 2016-04-08:
Ok no worried, Try this http://download.ardupilot.org/Copter.epub
After I figure it all we can add those to autocreation (weekly)

hamishwillee, 2016-04-08:
The requested URL /Copter.epub was not found on this server.

jpkh, 2016-04-08:
Ooops sorry.. http://download.ardupilot.org/epub/Copter.epub :)

hamishwillee, 2016-04-11:
Hi @jpkh 

I tried this on calibre viewer (Windows) and Kindle Paperwhite (converted via Kindle). In both cases the content worked fine. That is quite good because presumably you'd have had to update the youtube plugin to get it to build?

Anyway, the main flaw (on both platforms) is that the menu does not properly get generated which makes this of limited use since navigation is impossible. I think we should publish as-is with that as a known issue, and create a bug report for further investigation. Thoughts?

jpkh, 2016-04-25:
Yeah that sounds ok to me, i will make a system script that runs trough subfolders and generates all document bases

gmorph, 2016-11-15:
Is this something we really need?  The Wiki is changing all the time and to have a static ePub document that will be out of date very quickly I think is a mistake.  I think we are creating unnecessary work for ourselves with this and would prefer we didn't create an ePub document.

hamishwillee, 2016-11-16:
I don't think we strictly speaking "need" this, but it is a "nice to have". I certainly think a lot of the wiki remains useful over many months, and it is handy to have an offline version. Happy for @jpkh to make the call.

auturgy, 2017-10-15:
Any progress on this?  A wiki page to link to the ePubs would be good.

hamishwillee, 2017-10-15:
I very much doubt anything has been done. My last testing was that the epub built fine but was unnavigable due to missing TOC. Just building the "somewhat useful" epub would be easy, but should be automated.

brunoolivieri, 2019-08-06:
I do believe that it would be handy in the field configuration. 

ping  @jpkh. Did you the script for the whole wiki?

Hwurzburg, 2026-09-29:
8018 merged...closing

closed 2026-09-29 by Hwurzburg