ArduPilot/ardupilot_wiki #338: Build of single page as PDF

hamishwillee, 2016-03-29:


FLYBYME, 2016-04-13:
Yeah I would like this!

jpkh, 2016-05-07:
I will be really big pdf but shortly we should be able to do that. Sphinx allows it on some level

hamishwillee, 2016-05-07:
The really big PDF is trivially easy, but I think what people want to do is take away any arbitrary page offline.

gmorph, 2016-11-15:
They could just print it out of their browser to PDF - CTRL+P normally works.  I don't think there is a huge demand for this.

hamishwillee, 2016-11-16:
The PDF print from browser is rubbish because it includes lots of things you don't need in a PDF context (e.g. sidebar) and layout is bad. BUT I see this as "too hard" to do easily/not sufficient value for effort - and I agree the need isn't hug. 

So, closing for now.

closed 2016-11-16 by hamishwillee