ArduPilot/ardupilot_wiki #341: Fix broken links on copter

hamishwillee, 2016-03-30:
This is a list of all broken links detected on copter. Many are the missing items on firmware server. Note that not every broken link is necessarily permanently broken, but does indicate possible need to investigate

```
docs/channel-7-and-8-options.rst:166: [broken] http://firmware.ardupilot.org/downloads/wiki/pixhawk_sound_files/LostCopter.wav: HTTP Error 404: Not Found
docs/common-airbotpower-power-module.rst:7: [broken] http://www.airbotservices.com/airbotpower-pre-ordering-send-us-email: ''
docs/common-apm25-and-26-overview.rst:18: [broken] http://firmware.ardupilot.org/downloads/wiki/firmware/ArduCopter_APM_1.0_Firmware_3.2.1.zip: HTTP Error 404: Not Found
docs/common-apm25-and-26-overview.rst:18: [broken] http://firmware.ardupilot.org/downloads/wiki/firmware/ArduCopter_APM_2.0_Firmware_3.2.1.zip: HTTP Error 404: Not Found
docs/common-camera-shutter-triggering-for-sony-multiport-connectors-using-seagull-map.rst:36: [broken] http://www.seagulluav.com/seagull-map-support/: timed out
docs/common-camera-shutter-triggering-for-sony-multiport-connectors-using-seagull-map.rst:183: [broken] http://www.seagulluav.com/seagull-map-support/: timed out
docs/common-chdk-camera-control-tutorial.rst:62: [broken] http://www.brooxes.com/newsite/BBKK/BBKK-PARTS.html: HTTP Error 403: Forbidden
docs/common-chdk-camera-control-tutorial.rst:98: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/DM-Intervalometer.txt: HTTP Error 404: Not Found
docs/common-chdk-camera-control-tutorial.rst:324: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/3DR_Shoot.txt: HTTP Error 404: Not Found
docs/common-chdk-camera-control-tutorial.rst:404: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/3DR_Shoot.txt: HTTP Error 404: Not Found
docs/common-chdk-camera-control-tutorial.rst:574: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/3DRCHDKTester.txt: HTTP Error 404: Not Found
docs/common-choosing-a-ground-station.rst:186: [broken] http://fightingwalrus.com/: HTTP Error 404: Not Found
docs/common-choosing-a-ground-station.rst:257: [broken] http://fightingwalrus.com/: HTTP Error 404: Not Found
docs/common-choosing-a-ground-station.rst:208: [broken] https://groups.google.com/forum/#!forum/qgroundcontrol: Anchor '!forum/qgroundcontrol' not found
docs/common-connecting-the-radio-receiver-apm2.rst:20: [broken] http://dronefever.com/Jumper-cable-6-pin-to-6-pin-individual-female-15cm.html: timed out
docs/common-downloads_advanced_user_tools.rst:7: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:26: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ZYX-BMGC-EN_V1.5.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:27: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ZYX-BMGC-EN_V1.4.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:37: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ArduCopter_APM_1.0_Firmware_3.2.1.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:49: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/APMrover2-2.46.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:50: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/APMrover2-2.47.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:38: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ArduCopter_APM_2.0_Firmware_3.2.1.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:57: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/PPM-Encoder-V3-Manual.pdf: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:48: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/APMrover2-2.45.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:68: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/Mediatek_1.9_firmware-AXN1.51_2776_3329_384.1151100.5_v19.bin_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:85: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/USB_Driver_for_PX4_FMU_and_FLOW-PX4_Windows_Driver_2.0.0.4.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:96: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/3drradioconfig.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:118: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/GPS_UBLOX_passthrough_APM2.hex: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:106: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/3DR-Ublox.txt: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:129: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/Config_OSD_20.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:170: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ArduPPM_v2.3.16_ATMega32U2.hex_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:226: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ArduPPM_v2.3.16_ATMega328p_for_ArduCopter.hex_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:158: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/PPM-Encoder-3DR-Manual-v2.3.16.pdf: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:267: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/ArduPPM_v2.3.16_ATMega328p_for_ArduPlane.hex_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:362: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/MTK_DIYdrones.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:370: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/MinimOSD_20.hex_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:382: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/MinimOSD_22_hex.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:395: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/AP_OpticalFlow_test.hex: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:406: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/px4flow-klt-06Dec2014.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:417: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/APM_Arduino_Drivers.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:427: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/windows_parameter_file_for_flightgear_HIL_simulator_system.fgfsrc.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:24: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/hostapd_OdroidU3_Ubuntu1404.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:37: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/Airspeed_test.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:7: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:38: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/Airspeed_test1.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:48: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/ArduPilot-Arduino-1.0.3-Mac.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:56: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/ArduPilot-Arduino-1.0.3-windows.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:77: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/AttitudesThree.zip: HTTP Error 404: Not Found
[... cut ...]
docs/traditional-heli-parts-list.rst:13: [broken] http://store.3drobotics.com/products/df13-6-position-connector-45-cm: HTTP Error 404: Not Found
docs/xbee-radios.rst:207: [broken] http://forum.moltosenso.com/viewtopic.php?f16&t=8: HTTP Error 503: Service Unavailable
docs/xbee-radios.rst:207: [broken] http://www.moltosenso.com/#/pc=/client/fe/download.php: Anchor '/pc=/client/fe/download.php' not found
index.rst:164: [broken] https://groups.google.com/forum/#!forum/drones-discuss: Anchor '!forum/drones-discuss' not found
```

jpkh, 2016-03-30:
i should be able to fix most of those download files atleast

hamishwillee, 2016-03-30:
Fantastic.

jpkh, 2016-03-31:
A lot better now, few files need to dig from other sources...

docs/common-chdk-camera-control-tutorial.rst:574: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/3DRCHDKTester.txt: HTTP Error 404: Not Found
docs/common-chdk-camera-control-tutorial.rst:404: [broken] http://firmware.ardupilot.org/downloads/wiki/other_files/3DR_Shoot.txt: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:7: [broken] http://download.ardupilot.org/downloads/wiki/advanced_user_tools/: HTTP Error 403: Forbidden
docs/common-downloads_advanced_user_tools.rst:96: [broken] http://download.ardupilot.org/downloads/wiki/advanced_user_tools/3drradioconfig.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:68: [broken] http://download.ardupilot.org/downloads/wiki/advanced_user_tools/Mediatek_1.9_firmware-AXN1.51_2776_3329_384.1151100.5_v19.bin_.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:85: [broken] http://download.ardupilot.org/downloads/wiki/advanced_user_tools/USB_Driver_for_PX4_FMU_and_FLOW-PX4_Windows_Driver_2.0.0.4.zip: HTTP Error 404: Not Found
docs/common-downloads_advanced_user_tools.rst:427: [broken] http://download.ardupilot.org/downloads/wiki/advanced_user_tools/windows_parameter_file_for_flightgear_HIL_simulator_system.fgfsrc.zip: HTTP Error 404: Not Found
docs/common-downloads_developer_tools.rst:7: [broken] http://download.ardupilot.org/downloads/wiki/developer_tools/: HTTP Error 403: Forbidden
docs/common-downloads_developer_tools.rst:48: [broken] http://firmware.ardupilot.org/downloads/wiki/developer_tools/ArduPilot-Arduino-1.0.3-Mac.zip: HTTP Error 404: Not Found
docs/common-install-mission-planner.rst:25: [broken] http://www.microsoft.com/en-us/download/details.aspx?id=35: HTTP Error 503: Service Unavailable
docs/common-ppm-encoder-8-channel-standalone-encoder.rst:36: [broken] http://firmware.ardupilot.org/downloads/wiki/advanced_user_tools/PPM-Encoder-V3-Manual.pdf: HTTP Error 404: Not Found
docs/quick-start-guide.rst:40: [broken] http://www.microsoft.com/en-us/download/details.aspx?id=35: HTTP Error 503: Service Unavailable
docs/xbee-radios.rst:207: [broken] http://www.moltosenso.com/#/pc=/client/fe/download.php: Anchor '/pc=/client/fe/download.php' not found

jpkh, 2016-04-18:
Copter has now been fixed mostly. Some missing links still exists that points to old forums. Those will be fixed later when posts are activated on new DAPO server.

closed 2016-04-18 by jpkh

hamishwillee, 2016-04-19:
Thanks Jani - you're a star!

marcusealfa, 2016-05-25:
hamishwillee, I don't know how to help you better, but this link works:
http://www.moltosenso.com/#/pc==/client/fe/download.php

Bye.