ArduPilot/ardupilot_wiki #354: Adding a new MAVLink Message (Code Overview) is outdated

OXINARF, 2016-04-07:
[Adding a new MAVLink Message (Code Overview)](http://ardupilot.org/dev/docs/code-overview-adding-a-new-mavlink-message.html) - This page is outdated, we now have a mavlink submodule and generate.sh script doesn't exist anymore.

hamishwillee, 2016-04-07:
@OXINARF Thanks very much. Do you know enough about the actual process to make the updates? (can you help please?)

OXINARF, 2016-04-08:
@hamishwillee I'm terrible at documentation to be honest. The generation is now automatic in Ardupilot when compiling and adding a new message has now to be done in the MAVLink submodule. This change was made by @bugobliterator with help from @guludo, @lucasdemarchi and @tridge, if that is of any help.

hamishwillee, 2016-04-11:
@bugobliterator @lucasdemarchi @guludo Do you think you can help update this documentation? @OXINARF I can write the docs if I have enough information to try this myself - just basic bullet points on all the steps.

OXINARF, 2016-04-12:
@hamishwillee I'll try to compare with the steps that are there now:

1. mavproxy isn't needed to add messages to Ardupilot, although a GCS might be a good suggestion to test it in the end. Also, it may be good to say that git submodules should be updated and initialized (`git submodule update --recursive --init`)
2. Seems fine to me
3. Links are wrong, they are now at: https://github.com/ArduPilot/mavlink/tree/master/message_definitions/v1.0  But this is a submodule inside Ardupilot, so locally the XML files would be at _modules/mavlink/message_definitions/v1.0_. This means that locally the message should be added in the files at that path, but if proposing it for PR, it should go to the Ardupilot MAVLink repo (and probably upstream). Too complicated?
4. This step isn't needed, headers are automatically generated when building
5. Seems good
6. As I said in step 3, changes to the XML file go to the Ardupilot MAVLink repo and changes to vehicle/library code go to the Ardupilot repo. Also we no longer keep generated files in repos. And I'm not sure if you should continue to direct people to drones-discuss, probably we should change that to the Development category in the Discourse forum.

If there is any doubt I can try to clarify.

hamishwillee, 2016-04-14:
Hi @OXINARF 

Thanks very much. I'm just doing a new job so not got much time to review this yet. It does appear to make sense though. I'll make some changes and put in a PR for you to sanity check ... hopefully in the coming weeks.

magicrub, 2016-06-06:
I just fixed steps 3, 4 and 6. Still need to remove the drones discuss reference though.

auturgy, 2017-10-15:
Can someone work with @hamishwillee on this?  Good timing, as the mavlink.io guide is under re-write (by Hamish), so a joint effort would be good.

hamishwillee, 2017-10-15:
So to be very precise I'm working from Dronecode on a lot of things, including migrating relevant content from http://qgroundcontrol.org/mavlink/start to https://mavlink.io/en/ . As part of that work I hope to make sure that the content is accurate, useful, flight stack agnostic etc. Anyone who can help with that, please get in touch.

brunoolivieri, 2019-08-06:
I believe this issue is not specific, and the history for ardupilot_wiki/dev/source/docs/code-overview-adding-a-new-mavlink-message.rst shows some updates. 

I believe this issue could be closed.

Hwurzburg, 2019-11-02:
closing....stale issue

closed 2019-11-02 by Hwurzburg