ArduPilot/ardupilot_wiki #355: wiki build error

bys1123, 2016-04-07:
I was trying to make .pot file to upload to Transifex.
But I got a error like this when I use "make gettext"

here is the error log
```
# Sphinx version: 1.4
# Python version: 2.7.3 (CPython)
# Docutils version: 0.12 release
# Jinja2 version: 2.8
# Last messages:
#   done
#   writing output... [  1%] docs/3dr-kits-and-best-practices
#   writing output... [  2%] docs/ac2_followme
#   writing output... [  3%] docs/ac2_guidedmode
#   writing output... [  4%] docs/ac2_positionmode
#   writing output... [  5%] docs/ac2_simple_geofence
#   writing output... [  6%] docs/ac_rollpitchtuning
#   writing output... [  7%] docs/ac_throttlemid
#   writing output... [  8%] docs/ac_tipsfornewpilots
#   writing output... [  9%] docs/acro-mode
# Loaded extensions:
#   sphinx.ext.autodoc (1.4) from /usr/local/lib/python2.7/dist-packages/sphinx/ext/autodoc.pyc
#   sphinx.ext.intersphinx (1.4) from /usr/local/lib/python2.7/dist-packages/sphinx/ext/intersphinx.pyc
#   sphinxcontrib.youtube (unknown version) from /usr/local/lib/python2.7/dist-packages/sphinxcontrib_youtube-1.0dev_20160406-py2.7.egg/sphinxcontrib/youtube.pyc
#   alabaster (0.7.7) from /usr/local/lib/python2.7/dist-packages/alabaster/__init__.pyc
#   sphinx.ext.pngmath (1.4) from /usr/local/lib/python2.7/dist-packages/sphinx/ext/pngmath.pyc
#   sphinx.ext.todo (1.4) from /usr/local/lib/python2.7/dist-packages/sphinx/ext/todo.pyc
#   sphinx.ext.ifconfig (1.4) from /usr/local/lib/python2.7/dist-packages/sphinx/ext/ifconfig.pyc
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/sphinx/cmdline.py", line 244, in main
    app.build(opts.force_all, filenames)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/application.py", line 287, in build
    self.builder.build_update()
  File "/usr/local/lib/python2.7/dist-packages/sphinx/builders/__init__.py", line 251, in build_update
    'out of date' % len(to_build))
  File "/usr/local/lib/python2.7/dist-packages/sphinx/builders/__init__.py", line 322, in build
    self.write(docnames, list(updated_docnames), method)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/builders/__init__.py", line 360, in write
    self._write_serial(sorted(docnames), warnings)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/builders/__init__.py", line 366, in _write_serial
    doctree = self.env.get_and_resolve_doctree(docname, self)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/environment.py", line 1242, in get_and_resolve_doctree
    self.resolve_references(doctree, docname, builder)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/environment.py", line 1512, in resolve_references
    self._warn_missing_reference(refdoc, typ, target, node, domain)
  File "/usr/local/lib/python2.7/dist-packages/sphinx/environment.py", line 1548, in _warn_missing_reference
    self.warn_node(msg % {'target': target}, node, type='ref', subtype=typ)
TypeError: _warn_node() got an unexpected keyword argument 'type'
```
I've tried CentOS 6.4, ubuntu 14.04, ubuntu@vagrant use your vagrantfile, it's all like this. I don't know where I was wrong.

Thank you.

bys1123, 2016-04-07:
"make html" is also not working

hamishwillee, 2016-04-07:
I don't know why this has broken. I'm debugging now.

I am currently looking at seeing whether transifex live will work OK for us on the new site.

bys1123, 2016-04-07:
Mostly translator visit transifex is very slow, they only just want download it and edit it offline. Even the transifex online editor is heavy for us. transifex live is too big for chinese network.
po file is easy to upload and download for us.

hamishwillee, 2016-04-07:
I understand. I am investigating all options. Transifex live first though because we already have a lot of translated strings so I might have this up and running quickly.

I am of course happy to ALSO work with .po files, and do what you need to make that work.

hamishwillee, 2016-04-07:
Hi @bys1123 

Please confirm this build now works (as per our other discussion).

I'm going to have to do some testing of the PO file approach. I'm not yet sure if it is better to work via github or transifex or both with these files. Either way, I'd like the Chinese strings to be part of a single ArduPilot project so that all translations in all languages live in the same place. Does that seem reasonable?

Some issues with using the PO approach:

1. We lose the "in context" translation of Transifex live
2. Won't necessarily get the translations from "common" shared across wikis - there will be separate PO files for each wiki. Might be worth thinking about tooling to auto-copy/share the common strings.
3. Changes in github will change the PO files. This will need to be handled somehow.
4. We have to work out where/how to publish translations - effectively in their own sites. That isn't a problem for you of course, but it is for me :-)

bys1123, 2016-04-07:
Gov of china do lot of things make people apart form the world. Common people can't use Facebook, twitter. And gov try to make a new internet ecosystem in china. So, it's hard to use the same way like other country.
But I'd like to upload our translation on github in ArduPilot project.

Transifex has lots of good tool for translators, like matching or proofing. So we can't leave this powerful tool.
For common or same strings, if we translate it once. Transifex will match the others.

hamishwillee, 2016-04-07:
OK, thanks very much for your advice. I'll give those tools a go. And thanks for agreeing to upload the translation.

bys1123, 2016-04-07:
I'm very glad to see ardupilot better and better. So I'll do my best for it. If I could do something useful or if I did something wrong, let me know it.

bys1123, 2016-04-07:
Sphinx 1.3 works fine now.

bys1123, 2016-04-07:
I will upload our translations when we finished ten pages.

bys1123, 2016-04-07:
Yes, you are right, PO file lose the "in context".........

bys1123, 2016-04-07:
I'm trying use windows batch copy .rst to .txt and upload to Transifex. Seems like a easy way. 
"common" and each project still in its own folder.

hamishwillee, 2016-04-07:
I haven't tried this yet (hope to have time next week). I plan to follow these instructions http://docs.transifex.com/integrations/sphinx-doc/#file

bys1123, 2016-04-09:
https://www.transifex.com/droneplus/copter/content/
we put the English source file like this
[image]

bys1123, 2016-04-09:
And with click the “Auto update resources” button in the image above.
We can auto update resource from github
[image]

bys1123, 2016-04-14:
@hamishwillee Transifex can "Auto update resources" from github raw, I think maybe it's better than Transifex Live.

hamishwillee, 2016-04-18:
That sounds very cool. I haven't tried this yet sorry.

auturgy, 2017-10-15:
Bump.  Any news?  Is this resolved?

brunoolivieri, 2019-08-12:
I believe that it has not been under working anymore. 

As the multilanguage is not in the to-do list nowadays, and there is the CUAV Chinese initiative, I believe that we could close this issue.

auturgy, 2019-08-12:
Multilingual builds are definitely still on the radar. We just need to find a suitable way to do it in a sustainable way.

Regards,

James

Hwurzburg, 2019-11-02:
we have numerous issues on foreign language support in the wiki...this one is redundant...closing

closed 2019-11-02 by Hwurzburg