ArduPilot/ardupilot_wiki #357: Fixed Wing FAQ

govsux, 2016-04-08:
[Fixed Wing FAQ](http://ardupilot.org/plane/docs/fixed-wing-faq.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Suggestion 

It might be worthy to mention the throttle slew rate option under the 
'How do you stop a Nitro plane from cutting the engine in flight?'

I have had very good luck using this to avoid the engine 'gagging' and quitting in flight from the throttle being opening too fast which can be a common and annoying thing with a 2 stroke.

THR_SLEWRATE: Throttle slew rate   
maximum percentage change in throttle per second. A setting of 10 means to not change the throttle by more than 10% of the full throttle range in one second. Range: 0 127 Increment: 1 Units: Percent

Suggested verbiage....  THR_SLEWRATE Can also aid in prevention of a nitro engine stalling in flight by slowing the throttle advance from going wide open to quickly.

hamishwillee, 2016-04-08:
@govsux Thanks Rick. This doc is a bit crap/unmaintained. I think your suggestion makes sense.

@magicrub You ok with the suggested verbiage in the last sentence of https://github.com/ArduPilot/ardupilot_wiki/issues/357#issue-146783965 ?

gmorph, 2016-11-15:
I've updated this page with the suggestion.
Thanks, Grant.

closed 2016-11-15 by gmorph