ArduPilot/ardupilot_wiki #358: Add instructions on using Sphinx for docs natively

hamishwillee, 2016-04-08:
As some people do not want to use Vagrant. This is probably just a link to the Sphinx instructions.

benenaes, 2016-06-12:
Maybe it's an idea to advise to use virtualenv to make sure that the Python environment used for Sphinx doesn't break other things and/or that other Python upgrades don't break Sphinx ?

hamishwillee, 2016-06-12:
@benenaes In theory that is "always" a good idea but I really don't think it necessary - there is no evidence and my experience seems to be that Sphinx is rock solid and doesn't break other things. I have recommended our Vagrant installation because it is so very easy.

auturgy, 2017-10-15:
Can we close this?  I think if we start having native build env, we'll need Win, Mac, Linux etc etc etc, which just becomes difficult to maintain.  My suggestion is just having vagrant as the documented/supported build env, and not making life harder on ourselves.

hamishwillee, 2017-10-15:
As per top comment, this is probably worth doing - just by providing this link for anyone who wants native installation: http://www.sphinx-doc.org/en/stable/install.html

auturgy, 2017-10-16:
Closed by https://github.com/ArduPilot/ardupilot_wiki/pull/978

closed 2017-10-16 by auturgy

Ryanf55, 2024-08-01:
flake8's `--user` install breaks ROS 2 setups that rely on flake8 because it pins a user version of a package different than other tools need.