ArduPilot/ardupilot_wiki #361: RC Transmitter Flight Mode Configuration

dtjw315, 2016-04-08:
[RC Transmitter Flight Mode Configuration](http://ardupilot.org/copter/docs/common-rc-transmitter-flight-mode-configuration.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I created a blog post on DIYDRONES.COM for setting flight modes on the FLYSKY FS-I6 and figured it could be added to the transmitter list here at ardu pilot. Thanks.

http://diydrones.com/profiles/blogs/flysky-fs-i6-flight-modes

CraigElder, 2016-04-10:
Thank you for your interest. Please create a pull request with your changes to the wiki and they can be merged.

dtjw315, 2016-04-10:
Hi. Not sure how to create a 'pull request', I used the suggested changes link at the bottom of the relevant page.

CraigElder, 2016-04-10:
Right up at the top of the wiki page is a link that says edit in Github. Click the link, make your edits, create a pull request. 
Also reference http://ardupilot.org/copter/docs/common-wiki_editing_guide.html

hamishwillee, 2016-04-11:
As @CraigElder says, it's pretty easy to make changes. This is [quick edit](http://ardupilot.org/copter/docs/common-wiki_editing_guide.html#making-a-quick-edit) so Github pretty much guides you on how to make a Pull Request (also known as a PR). 

Thanks very much for this link - we'd certainly like to have it!

dtjw315, 2016-04-11:
Ok guys, I clicked on the 'edit on github' link, but it takes me to a 404 error page.

hamishwillee, 2016-04-13:
@dtjw315 Thanks - you can use this link: https://github.com/ArduPilot/ardupilot_wiki/blob/master/common/source/docs/common-rc-transmitter-flight-mode-configuration.rst

I've created #366 to fix the broken "Edit on Github" link.

hamishwillee, 2016-04-13:
And you should be able to click the Edit on Github link now too!

dtjw315, 2016-04-13:
Thanks, Hamish and yes, that opened fine now. I edited the page, clicked 'submist proposal' and then clicked pull request. Am new to this, so I hope I got it right.

hamishwillee, 2016-04-14:
@dtjw315 has addressed this with #376

hamishwillee, 2016-04-14:
HI @dtjw315 

Looks good! Now you've created that we basically work on #376 (ignoring this post) until it addresses this issue and then merge the PR. Once merged, the new version is the "live version" and we can close this issue. 

So now, wander over to #376 and see my question :-)

closed 2016-04-14 by hamishwillee