ArduPilot/ardupilot_wiki #363: Why add linefeed manually?

bys1123, 2016-04-09:
[image]

In transifex it's like this cauce linefeed

hamishwillee, 2016-04-13:
Hi @bys1123 

Sorry, I missed this message! 

Adding linefeeds makes it easier to read in the github editor and to see changes in the file comparison, because changes are compared on a line by line basis. I had not considered translation at all. Also, important to note that - anyone can contribute and they will sometimes do this. 

So what can we do about this? I'm happy for you to patch the source removing these line breaks, and to suggest in the wiki editing guide that people to not add artificial breaks. We/you can then make this easier "as we go along". Will that work for you?

bys1123, 2016-04-14:
Do you think one line one sentence is good?

hamishwillee, 2016-04-16:
Not really, because a sentence can have "any" length, and sentences should be joined in paragraphs to keep common ideas together.

Sphinx ignores the line breaks. The only reason to have them is to make reviewing change easier in Github.

bys1123, 2016-04-17:
Yeah, I mean one line one sentence in github. 
Easier to reviewing changes and translation. Sphinx ignores the line breaks, so sentences still joined in paragraphs.

hamishwillee, 2016-04-18:
Thanks for the explanation.

I don't think we should proscribe that lines **must** break on sentences, but I think it would be reasonable to *recommend* that they aren't broken mid sentence:

    Please do not add artificial line breaks in the middle of 
    sentences as this can make it harder for content translation.

This will make reviewing a little harder, but I think well worth it to make the translation job easier for you. Note though that as this is a wiki, we can't guarantee that authors will follow this rule, and you may need to patch articles. 

Make sense?

bys1123, 2016-04-22:
@hamishwillee 
I've watched https://github.com/PX4/Devguide , I think it's better than Sphinx.
Is there any chance to transfor the wiki form Sphinx to GitBook,  Can pandoc do this?
Transifex can read the md files, then no linefeed problem.

hamishwillee, 2016-04-23:
Hi @bys1123 

The PX4 devguide is based on Gitbook, which is nowhere near as good as Sphinx. It's quite limited in terms of managing links, but even worse it only supports 2 levels deep in the sidebar as part of its core implementation - which is not enough for ArduPilot. 

Markdown itself is fine, but the linking it supports is purely based on relative paths, which means that if anything moves links get broken. This wouldn't be a particular problem now, but might if we change things in future - Sphinx copes with this case really well.  Sphinx does support Markdown, but we don't use it because to me having links work well is very important for maintenance.

We could move to a markdown system relatively easily with the above caveats, and still use Sphinx. It really depends on whether the dev team wants to make that change. I doubt it, but I will ask.

bys1123, 2016-04-27:
@hamishwillee 
OK, thanks for explain. It's seems like recommend that authors aren't broken mid sentence is the best way. 
We can help you to patch articles at the beginning of next month. Our new translator team is more powerful now. After the Dronecode Unconference Beijing today, I think our team is gonna be lots of better and bigger.

gmorph, 2016-11-15:
Is this still an open issue or can we close this out?

hamishwillee, 2016-11-16:
I think we can close this. Stone has been quiet so either they aren't translating or this isn't a problem. Either way, I wouldn't proceed this without a "stakeholder".

closed 2016-11-16 by hamishwillee

bys1123, 2016-12-23:
We are developing a web translating tool.