ArduPilot/ardupilot_wiki #368: Complete Parameter List

trondhindenes, 2016-04-11:
[Complete Parameter List](http://ardupilot.org/copter/docs/parameters.html) - It would be great if the default values were provided here for reference, and if the list was ordered as it is in mission planner

hamishwillee, 2016-04-13:
@peterbarker Can we/do we provide default values. 

@trondhindenes I think the ordering makes sense. Mission Planner is not the only GCS

peterbarker, 2016-04-13:
On Tue, 12 Apr 2016, Hamish Willee wrote:

> @peterbarker Can we/do we provide default values.

This was discussed in the last dev call.  It is not possible tell from the 
source code what the default values are; the code has to be run to 
determine that.  Considering the variety of platforms we run on (and the 
fact that those platforms will get different default values), I don't 
think we can supply a single e.g. Parameters.rst which has "default" 
values in it.

hamishwillee, 2016-04-13:
Hi @trondhindenes 

I am closing this as "wontfix". Though it might be nice to have default values it isn't easy to do as peter points out. I'm not even sure it is useful, since the default is typically loaded from a separate file, and will depend on the specific vehicle.

closed 2016-04-13 by hamishwillee