ArduPilot/ardupilot_wiki #372: Long words don't wrap onto multiple lines, and page can not be panned right.

WickedShell, 2016-04-11:
Long descriptions like the last word of http://ardupilot.org/plane/docs/parameters.html#log-bitmask-log-bitmask doesn't fit on a single screen width, and the last part of the bit description is not readable.

hamishwillee, 2016-04-13:
@WickedShell Thanks, this is a known issues.

@peterbarker Your latest update to the parameters on plane fixes this rather nicely. Did you submit the patch?

peterbarker, 2016-04-13:
On Tue, 12 Apr 2016, Hamish Willee wrote:

> @WickedShell Thanks, this is a known issues.
> 
> @peterbarker Your latest update to the parameters on plane fixes this rather
> nicely. Did you submit the patch?

Not yet.  Soon.

brunoolivieri, 2019-08-12:
It seems like the words in parameters file are wrapping correctly in multiple lines now. Right?

If so, I believe that this issue could be closed.

Hwurzburg, 2019-08-12:
Confirmed..closing

closed 2019-08-12 by Hwurzburg