ArduPilot/ardupilot_wiki #373: UBlox GPS Configuration

moon-siama, 2016-04-12:
There is a wrong configuration in the 3DR-Ublox_NEO7.txt

CFG-NAV5 -- In the Navigation Modes, default setting of Dynamic Model is Portable, which should be Airborne < 4g

Others are setting to Airborne < 4g

I would like to know there is any new changes if using NEO7 module

Thanks

rmackay9, 2016-04-12:
@moon-siama, 
You don't happen to know what exactly should be in the file for the NAV5 line do you?

moon-siama, 2016-04-12:
Thanks Randy for your reply.

Here is my understanding,

<3DR-Ublox.txt>
CFG-NAV5 - 06 24 24 00 FF FF **08** 03 00 00 00 00 10 27 00 00 05 00 FA 00 FA 00 64 00 2C 01 00 3C 00 00 00 00 00 00 00 00 00 00 00 00

31.10.2 Get/Set Navigation Engine Settings
Message CFG-NAV5
Description Get/Set Navigation Engine Settings
Firmware Supported on u-blox 6 from firmware version 6.00 up to version 7.03.
Type Get/Set
Message Structure 0xB5 0x62 0x06 0x24 36 see below CK_A CK_B

**2 U1 - dynModel - Dynamic Platform model:**
- 0 􀀀 Portable
- 2 􀀀 Stationary
- 3 􀀀 Pedestrian
- 4 􀀀 Automotive
- 5 􀀀 Sea
- 6 􀀀 Airborne with <1g Acceleration
- 7 􀀀 Airborne with <2g Acceleration
**- 8 􀀀 Airborne with <4g Acceleration**

**Source from u-blox 6 Receiver Description GPS.G6-SW-10018-F (Revision for FW 7.03)**

<3DR-Ublox_NEO7.txt>
CFG-NAV5 - 06 24 24 00 FF FF **00 ** 03 00 00 00 00 10 27 00 00 05 00 FA 00 FA 00 64 00 2C 01 00 3C 00 00 00 00 00 00 00 00 00 00 00 00

35.9.2 Navigation Engine Settings
Message CFG-NAV5
Description Navigation Engine Settings
Firmware Supported on:• u-blox 7 firmware version 1.00
Type Input/Output
Header ID Length (Bytes) Payload Checksum
Message Structure 0xB5 0x62 0x06 0x24 36 see below CK_A CK_B

**2 U1 - dynModel - Dynamic Platform model:**
**0 Portable**
2 Stationary
3 Pedestrian
4 Automotive
5 Sea
6 Airborne with <1g Acceleration
7 Airborne with <2g Acceleration
8 Airborne with <4g Acceleration

**Source from . u-blox 7 Receiver Description GPS.G7-SW-12001-B (Protocol version 14.00)**

<Macro-Ublox_M8N.txt>
CFG-NAV5 - 06 24 24 00 FF FF **08** 03 00 00 00 00 10 27 00 00 05 00 FA 00 FA 00 64 00 2C 01 00 3C 00 00 00 00 C8 00 00 00 00 00 00 00

27.11.14.1 Navigation Engine Settings
Message CFG-NAV5
Description Navigation Engine Settings
Firmware Supported on:
• u-blox 8 / u-blox M8 from protocol version 15 up to version 22
Type Get/Set
Header Class ID Length (Bytes) Payload Checksum
Message Structure 0xB5 0x62 0x06 0x24 36 see below CK_A CK_B

**2 U1 - dynModel - Dynamic platform model:**
0: portable
2: stationary
3: pedestrian
4: automotive
5: sea
6: airborne with <1g acceleration
7: airborne with <2g acceleration
**8: airborne with <4g acceleration**
9: wrist worn watch (not supported in protocol
versions less than 18)

**Source from . u-blox 8 / u-blox M8 Receiver Description**

Actually, I did test Loiter mode with Neo7 module. But it didn't work, it was always glitched and gave a wrong location info during flight time.

And I did test Loiter mode with LEA 6H module and result was fine, Loiter working was good.

So, I think you are better than me, could you please give me your idea to solve this problem?

Thanks

magicrub, 2016-04-13:
@WickedShell can you take a look at this?

WickedShell, 2016-04-14:
Thanks for tagging me @magicrub I hadn't seen this at all. (I should probably watch wiki issues).

Looks reasonable, I still need to confirm with u-center for the file output. But yes you want to be using 4G mode.

What firmware version are you flying? (plane/copter?)  The reason I'm asking is the autopilot attempts to configured the nav mode based on a parameter GPS_NAVFILTER which defaults to 4G as well. If you are using the newest version of Plane it even goes as far as checking that the field is set to the correct value as one of the arming checks, so I would have expected this to not be a problem in flight.

I haven't ever looked at either of those files that are on the wiki at the moment as I allow the autopilot to do all my configuration. I will try and review them sometime this evening or tomorrow. I think there should maybe be a discussion about moving away from those files and allowing the autopilot to do all the configuration (including optionally saving it) since it now verifies that it set the value correctly, and @magicrub added support for only saving if we made a change (rather then saving all the time). One of the problems with these files is that a lot of people don't realize that the AP makes configuration choices at run time depending on parameters. (There is a flag to opt out of allowing the autopilot to make these configuration changes GPS_AUTOC_CONFIG).

moon-siama, 2016-04-14:
Thanks for your reply, @WickedShell

Yes, it is saved configuration of GPS during AP_GPS_UBLOX init (I am not sure), I have tested with Copter 3.3 git version and also master git version

But Copter 3.3 has only GPS_NAVFILTER paramter.
And also I am not sure that 
  - if Pixhawk has been initialized which mean is powered on, CFG_NAV5 setting was G4 (I confirmed it with Mission Planner and Pixhawk as Passthrough Method) and after power off and directly connect the GPS module to U-Center, CFG_NAV5 setting was back to Portable. ( I think it may need more command message such as GPS_SAVE_CFG parameter in master.

 I think those procedures are very simple and easy to change with parameters.

But  if Ublox gps module with the manufacturing setting or default setting, that procedure may be not covered other configuration settings (if other settings are essential), so I think  someone like you have a look at the** 3DR_Ublox_NEO7.txt file **

Thanks

WickedShell, 2016-04-14:
If you power cycle and haven't set GPS_SAVE_CFG to 1 or 2 then it won't save it on the GPS. (The default behavior of the autopilot is to set reasonable defaults, but do not force save them unless requested by the user (by changing that SAVE_CFG parameter). If you were to connect with the pass through option after the GPS has been configured you should see it set to 4G.

You probably new this already since its documented on the parameters, but any changes you make to the GPS_ parameters tend to need a reboot to apply the changes.

closed 2016-11-15 by davidbuzz