ArduPilot/ardupilot_wiki #381: SingleCopter and CoaxCopter Support

ChrisKrieg, 2016-04-15:
[SingleCopter and CoaxCopter](http://ardupilot.org/copter/docs/singlecopter-and-coaxcopter.html) - SingleCopter and CoaxCopter still not supported in ardupilot GUI

Is there a timeline when a firmware build for these frame will become available for download as described at the end of the article? I have had extreme difficulty trying to build the profile using arduino for both APM2.6 and Pixhawk. It seems that support for these frame types has died or gone deep underground. Is there any interest in supporting them more fully in the future?

Thanks.

rmackay9, 2016-04-15:
@ChrisKrieg, yes, there's definitely interest.  One problem we have is that nobody on the dev team has a frame to test with.
In particular, in Copter-3.4 we've done some changes that we worry will break single copter and coax copter so we need someone to test it.  We could provide you with a binary for a pixhawk if you have a frame and could help us test.  Sound ok?  If yes, can you email drones-discuss@googlegroups.com or create a new discussion on diydrones.com (and link it here) so you, me and Leonard can talk?

rmackay9, 2016-04-15:
i'm going to close this issue because it's not really related to the wiki.

closed 2016-04-15 by rmackay9

ChrisKrieg, 2016-04-18:
Do you want me to start a new topic on http://discuss.ardupilot.org/c/arducopter/3-3 ? That seems to be the new support forum.

rmackay9, 2016-04-18:
@ChrisKrieg, I would just create a new topic under "ArduCopter".  http://discuss.ardupilot.org/c/arducopter

There's actually another guy there interested in building a single copter it seems.

ChrisKrieg, 2016-04-19:
Here's the link to the new thread http://discuss.ardupilot.org/t/singlecopter-coaxcopter-support/8543