ArduPilot/ardupilot_wiki #388: Seagull #REC Trigger

Fireycold, 2016-04-18:
The Seagull #REC should be added to the list of compatible triggers for Pixhawk and Sony cameras.
The Seagull #REC adds additional features to those offered by its predecessor, the #MAP 2.  The manufacturer's website includes details on setup with the Pixhawk system.
http://www.seagulluav.com/product/seagull-rec/
http://www.seagulluav.com/manuals/Seagull_REC-Manual.pdf

hamishwillee, 2016-04-18:
@Fireycold That would be nice - can you [create a PR](http://copter.ardupilot.org/ardupilot/docs/common-wiki_editing_guide.html) with the needed change?

You would need to add [a section here](https://github.com/ArduPilot/ardupilot_wiki/blob/master/common/source/docs/common-camera-shutter-triggering-cables-for-purchase.rst). YOu might also add a dedicated document like this one for [Seagull #Map](http://ardupilot.org/copter/docs/common-camera-shutter-triggering-for-sony-multiport-connectors-using-seagull-map.html) (but that would be optional).

davidbuzz, 2016-11-15:
wiki page changed from 'MAP' to 'REC' with updated manual and product links.

closed 2016-11-15 by davidbuzz

hamishwillee, 2016-11-15:
Works for me.