ArduPilot/ardupilot_wiki #391: Building ArduPilot for APM2.x with Eclipse on Windows

liaorongfan, 2016-04-22:
[Building ArduPilot for APM2.x with Eclipse on Windows](http://ardupilot.org/dev/docs/building-apm2-with-eclipse-on-windows.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

closed 2016-04-22 by liaorongfan

rmackay9, 2016-04-22:
we should totally remove this wiki page actually.  We don't support the APM2.x boards anymore.

liaorongfan, 2016-04-22:
How about the earlier branches of codes for ArduCopter based on an APM2.X,such as V2.8 or V2.5, I mean ,does it works for building on Arduino1.0.3? Looking forward to hearing from you. Thank you!

发自我的 Kevin

rmackay9, 2016-04-23:
@liaorongfan, yes, the instructions should still work for very old builds.

hamishwillee, 2016-04-24:
FYI @rmackay9 I've archived this as #406

rmackay9, 2016-04-25:
great, thanks!

liaorongfan, 2016-04-25:
@rmackay9  Got it, thanks a lot

liaorongfan, 2016-05-05:
@rmackay9 Excuse me, I am still working on Building ArduPilot for APM2.x with Eclipse on Windows,since the artical says "They have been tested on Windows 10 to build the ArduCopter-3.2.1 branch."  But after I type "make apm2" in the px4 console,it responded "Can not find Arduino in this system", though I specified the path in config.mk  like this "ARDUINO=C:/APM_program/ArduPilot-Arduino-1.0.3-windows". could you give me some advice?

rmackay9, 2016-05-05:
@liaorongfan, I'm sorry, I just can't keep supporting the AVR boards.  I get so many requests each day for support and new features, I'm afraid I have to prioritise those that help move the project forward.